LibreHardwareMonitor/LibreHardwareMonitor #906: [Feature request] Add support for installation via winget

DerekTurtleRoe, 2022-11-24:
You can learn more about winget [here](https://github.com/microsoft/winget-cli).

Opening a pull request with the package details [here](https://github.com/microsoft/winget-pkgs) will add it to the winget repositories and enable support. Being able to support installing this via the command-line and/or scripts would be really handy for me! 😄 

You can use several scripts and generators to create the needed information, available [here](https://github.com/microsoft/winget-pkgs#authoring-a-manifest).

I have included an example below, which should be most of the needed info for the PR:

[LibreHardwareMonitor-winget.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10080964/LibreHardwareMonitor-winget.zip)

jo-chemla, 2022-12-19:
Just added a link in [this issue](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/903) which suggests adding a chocolatey package, which can also be useful for the winget package creation: how dolt uses [github actions](https://www.dolthub.com/blog/2021-10-13-how-to-publish-winget-chocolatey-packages-with-github-actions/) to automate the package deployment.

eabase, 2023-06-23:
BUMP! 

I would love to see this in winget.
I don't want to have to install yet another package manager to have this.

nlxdodge, 2025-02-14:
I am not sure who did, but there are two versions already: [GitHub Files](https://github.com/microsoft/winget-pkgs/tree/master/manifests/l/LibreHardwareMonitor/LibreHardwareMonitor)

Not sure if someone automated it, but for now it's at the latest version.

eabase, 2025-02-16:
Cool. Maybe add that to the README.

AMDphreak, 2025-03-03:
The package's descriptive name needs to be modified to "Libre Hardware Monitor" for search purposes, shown below:
[image]

Also, someone should delete the deprecated chocolatey package. Its status as deprecated does not show up in winget or UniGetUI

[image]

nlxdodge, 2025-03-10:
I checked the original PR to [here](https://github.com/microsoft/winget-pkgs/pull/193535)

It seems like [@a-mnich](https://github.com/a-mnich) and [@Utesgui](https://github.com/Utesgui) created a bot that automatically updates software?

Can you guys maybe help on this? Did you already automate the PR creation every time LibreHardwareMonitor gets an update?

If that is the case, then we only need to update the text in the PR, and remove/depricate chocolatey. I can maybe check that later tonight.

nlxdodge, 2025-03-11:
@eabase created a [PR](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1659) for the Readme, if anything needs to be changed, let me know in that PR 👍🏻

PhyxionNL, 2025-06-13:
Cheers, closing.

closed 2025-06-13 by PhyxionNL

jo-chemla, 2025-07-01:
You could simply disable the chocolatey source from UnigetUI - all the ~100 packages I use have migrated to winget. 
Regarding search, the closest issue I could find from your search spaces-vs-dashes is https://github.com/microsoft/winget-cli/issues/1966 if you want to watch it

AMDphreak, 2025-07-02:
> You could simply disable the chocolatey source from UnigetUI - all the ~100 packages I use have migrated to winget. Regarding search, the closest issue I could find from your search spaces-vs-dashes is [microsoft/winget-cli#1966](https://github.com/microsoft/winget-cli/issues/1966) if you want to watch it

That's not addressing the point. I do frequently disable chocolatey, but when comparing apples to apples, the choco outdated package still looks newer than both the other choco pkg and winget pkg.