LibreHardwareMonitor/LibreHardwareMonitor #909: Remote Server can not acceess from other devices

seanmars, 2022-11-24:
I run the remote server but can't access it from other devices. I have tried to disable the firewall but it can't access too.

ntoff, 2022-12-05:
>I have tried to disable the firewall

I can't reproduce this. Are both of your computers on the same network?

For me:
windows firewall **enabled**: no access to LWHM web server
windows firewall **disabled**: LHWM web server is accessible
windows firewall **enabled AND port opened**: LHWM server is accessible

If you're going to leave the firewall enabled, but just open the one port it uses, it seems that LHWM's exe isn't the listener, but "system" (i.e. windows) is. I had to open the port the server listens on to "any application" which isn't ideal, but on an internal network behind a firewall probably not so much of an issue.

seanmars, 2022-12-06:
> I can't reproduce this. Are both of your computers on the same network?
>
> For me: windows firewall **enabled**: no access to LWHM web server windows firewall **disabled**: LHWM web server is [...]

I'm on the same network and have tried to disable the firewall. But it's still can not accessible.

Tar-rask, 2023-04-09:
@seanmars  Did you ever solve this? I seem to have a similar problem. I was running openhardware monitor fine. But some reason I can't connect to the web server remotely, even after turning off the firewall (windows 10)