LibreHardwareMonitor/LibreHardwareMonitor #924: Clevo based notebooks fan control

uterus-blower, 2022-12-09:
I tried to use FanControl ui with LibreHardwareMonitor but seems like it doesn't support Clevo based laptops. I don't know much about coding if you guys can add support it would be awesome. 

If it helps I tried some of these apps, they can control fan speed but they are so basic.

[image]

Apps have support:

https://github.com/zuyan9/RLECViewer (tested)

https://code.obsidian-pc.com/clevo-software/fan-control/ (paid)

https://github.com/tuxedocomputers/tuxedo-fan-control (for linux)

mikeloeven, 2022-12-18:
Same here Clevo is one of the larger manufacturers of laptop chassis for smaller brands adding Clevo support would open up A LOT of off brand laptops that are not otherwise properly supported by fan control software

mikhailmalashchenko, 2023-09-21:
Are there any plans to add support for Clevo laptops, specifically the x170km-g, to your software?