LibreHardwareMonitor/LibreHardwareMonitor #930: GpuNvidia is missing in Windows Forms Application (C#)

CaptainAlekor, 2022-12-13:
GpuNvidia is missing in Windows Forms Application (C#). On a screenshot i ran same code, but one in console app and the other in Windows Forms application (Visual Studio, C#). You can notice, that there are no "Hardware: NVIDIA GeForce RTX 3050 Ti Laptop GPU" in Windows Forms app. What is the problem, how can i fix it?
[image]

CaptainAlekor, 2022-12-13:
Solved: setting solution platform to x64 fixed the problem.
[image]

closed 2022-12-13 by CaptainAlekor

JunSourceX, 2023-02-03:
> Solved: setting solution platform to x64 fixed the problem. [image]

Thank you, you helped me)