LibreHardwareMonitor/LibreHardwareMonitor #931: Win 11: "System Tray" icon somehow garbled, "pixeled"

ralfbergs, 2022-12-17:
The "System Tray" temperature display "icon" appears somehow garbled, "pixeled"

[image]

I tried to find any setting that could improve the visualization, but it seems this is just some kind of bug?

Before I upgraded to Win11, the very same app version behaved properly, i. e. temp display was very crispy and crystal clear.

Running latest build 0.9.1 under Win11, build 22H2 (22621.963).

playalisticp, 2023-06-09:
I have the exact same issue, but on Windows 10 22H2. Actually, it was never fine on my machine, has been always pixelated.
PC from my uncle has this too, Windows 11.

akaspeh1, 2024-01-21:
#1249 
May be related to the code drawing the try icon on transparent background.
Ported the code from the working openHW monitor in the pull request. 
Hopefully it will fix it for you.

closed 2024-08-26 by PhyxionNL