LibreHardwareMonitor/LibreHardwareMonitor #936: CPU Temperatures not available in C# application

Tretgo, 2022-12-22:
Hi,
I am using a 13600k on a Gigabyte Z690 Aorus Elite DDR4.
I can see core temperatures in the GUI, but they in C# they are always reported as 0. Mainly I am interested in Core Max value.
GPU Temperature for example works.

Code is the example from the readme.

C# code output
[image]
Monitor GUI
[image]

btrautmann, 2023-01-10:
FWIW I'm seeing the same on a 13700k on an MSI Pro Z690A. Here's a JSON output of what I'm receiving using the sample code with same _slight_ modifications for added clarity:

```json
{
  "MSI PRO Z690-A (MS-7D25)": {
    "type": "Motherboard"
  },
  "13th Gen Intel Core i7-13700K": {
    "type": "Cpu",
    "CPU Core #1 Thread #1-Load": "9.07169",
    "CPU Core #1 Thread #2-Load": "2.176869",
    "CPU Core #2 Thread #1-Load": "0",
    "CPU Core #2 Thread #2-Load": "0",
    "CPU Core #3 Thread #1-Load": "0",
    "CPU Core #3 Thread #2-Load": "1.7320693",
    "CPU Core #4 Thread #1-Load": "0",
    "CPU Core #4 Thread #2-Load": "0",
    "CPU Core #5 Thread #1-Load": "54.191486",
    "CPU Core #5 Thread #2-Load": "0",
    "CPU Core #6 Thread #1-Load": "58.640564",
    "CPU Core #6 Thread #2-Load": "0",
    "CPU Core #7 Thread #1-Load": "0",
    "CPU Core #7 Thread #2-Load": "0",
    "CPU Core #8 Thread #1-Load": "0",
    "CPU Core #8 Thread #2-Load": "0",
    "CPU Core #9-Load": "12.43856",
    "CPU Core #10-Load": "11.260104",
    "CPU Core #11-Load": "4.365653",
    "CPU Core #12-Load": "5.1221313",
    "CPU Core #13-Load": "13.625706",
    "CPU Core #14-Load": "7.5961056",
    "CPU Core #15-Load": "5.124241",
    "CPU Core #16-Load": "9.545732",
    "CPU Total-Load": "6.152606",
    "CPU Core #1-Temperature": "",
    "CPU Core #2-Temperature": "",
    "CPU Core #3-Temperature": "",
    "CPU Core #4-Temperature": "",
    "CPU Core #5-Temperature": "",
    "CPU Core #6-Temperature": "",
    "CPU Core #7-Temperature": "",
    "CPU Core #8-Temperature": "",
    "CPU Core #9-Temperature": "",
    "CPU Core #10-Temperature": "",
    "CPU Core #11-Temperature": "",
    "CPU Core #12-Temperature": "",
    "CPU Core #13-Temperature": "",
    "CPU Core #14-Temperature": "",
    "CPU Core #15-Temperature": "",
    "CPU Core #16-Temperature": "",
    "CPU Package-Temperature": "",
    "CPU Core #1 Distance to TjMax-Temperature": "",
    "CPU Core #2 Distance to TjMax-Temperature": "",
    "CPU Core #3 Distance to TjMax-Temperature": "",
    "CPU Core #4 Distance to TjMax-Temperature": "",
    "CPU Core #5 Distance to TjMax-Temperature": "",
    "CPU Core #6 Distance to TjMax-Temperature": "",
    "CPU Core #7 Distance to TjMax-Temperature": "",
    "CPU Core #8 Distance to TjMax-Temperature": "",
    "CPU Core #9 Distance to TjMax-Temperature": "",
    "CPU Core #10 Distance to TjMax-Temperature": "",
    "CPU Core #11 Distance to TjMax-Temperature": "",
    "CPU Core #12 Distance to TjMax-Temperature": "",
    "CPU Core #13 Distance to TjMax-Temperature": "",
    "CPU Core #14 Distance to TjMax-Temperature": "",
    "CPU Core #15 Distance to TjMax-Temperature": "",
    "CPU Core #16 Distance to TjMax-Temperature": "",
    "Core Max-Temperature": "",
    "Core Average-Temperature": "",
    "CPU Core #1-Clock": "",
    "CPU Core #2-Clock": "",
    "CPU Core #3-Clock": "",
    "CPU Core #4-Clock": "",
    "CPU Core #5-Clock": "",
    "CPU Core #6-Clock": "",
    "CPU Core #7-Clock": "",
    "CPU Core #8-Clock": "",
    "CPU Core #9-Clock": "",
    "CPU Core #10-Clock": "",
    "CPU Core #11-Clock": "",
    "CPU Core #12-Clock": "",
    "CPU Core #13-Clock": "",
    "CPU Core #14-Clock": "",
    "CPU Core #15-Clock": "",
    "CPU Core #16-Clock": ""
  },
  "Generic Memory": {
    "type": "Memory",
    "Memory Used-Data": "16.188053",
    "Memory Available-Data": "15.604786",
    "Memory-Load": "50.917294",
    "Virtual Memory Used-Data": "29.536663",
    "Virtual Memory Available-Data": "7.006176",
    "Virtual Memory-Load": "80.8275"
  },
  "NVIDIA GeForce RTX 3080": {
    "type": "GpuNvidia",
    "GPU Core-Temperature": "53",
    "GPU Core-Clock": "210",
    "GPU Memory-Clock": "405",
    "GPU Fan 1-Fan": "0",
    "GPU Fan 2-Fan": "0",
    "GPU Fan 1-Control": "0",
    "GPU Fan 2-Control": "0",
    "GPU Core-Load": "5",
    "GPU Memory Controller-Load": "19",
    "GPU Video Engine-Load": "0",
    "GPU Bus-Load": "0",
    "GPU Power-Load": "9.045",
    "GPU Board Power-Load": "15.852",
    "D3D Dedicated Memory Used-SmallData": "1420.4805",
    "D3D Shared Memory Used-SmallData": "109.96094",
    "D3D 3D-Load": "5.4975805",
    "D3D Compute_0-Load": "0",
    "D3D Overlay-Load": "0",
    "D3D Video Decode-Load": "0",
    "D3D Copy-Load": "0.05035717",
    "D3D Security-Load": "0",
    "D3D Video Encode-Load": "0",
    "D3D Video Processing-Load": "0",
    "D3D Graphics_1-Load": "0",
    "D3D Cuda-Load": "0",
    "D3D Compute_1-Load": "0",
    "D3D VR-Load": "0",
    "GPU Hot Spot-Temperature": "59.3125",
    "GPU Memory Junction-Temperature": "56",
    "GPU Memory Total-SmallData": "12288",
    "GPU Memory Free-SmallData": "10694",
    "GPU Memory Used-SmallData": "1593",
    "GPU Memory-Load": "12.964535",
    "GPU Package-Power": "32.279",
    "GPU PCIe Rx-Throughput": "72704000",
    "GPU PCIe Tx-Throughput": "6144000"
  },
  "Intel(R) UHD Graphics": {
    "type": "GpuIntel",
    "D3D Shared Memory Used-SmallData": "2.15625",
    "D3D 3D-Load": "0",
    "D3D Video Decode-Load": "0",
    "D3D Copy-Load": "0",
    "D3D Video Processing-Load": "0",
    "D3D Compute-Load": "0",
    "D3D Other-Load": "0",
    "D3D GDI Render-Load": "0"
  },
  "Ethernet": {
    "type": "Network",
    "Data Uploaded-Data": "0.07547829",
    "Data Downloaded-Data": "3.5954487",
    "Upload Speed-Throughput": "592.58875",
    "Download Speed-Throughput": "1601.8414",
    "Network Utilization-Load": "0.0017555442"
  }
}
```

You can see that `Distance to TjMax-Temperature` and `Clock` are also not reported.

I'm just starting to nibble at the actual library code (I'm using it in a .NET MAUI project that broadcasts the data over a SignalR hub to be shown on a Flutter app), so I may find the issue eventually, but wanted to drop a +1 here in the meantime just in case.

Tretgo, 2023-01-11:
FYI I circumvented it by using the WMI interface.

`            Process pProcess = new Process();
            pProcess.StartInfo.FileName = "wmic";
            pProcess.StartInfo.Arguments = @"/namespace:\\root\LibreHardwareMonitor PATH Sensor where (Name='GPU Core' and SensorType='Temperature' or Name='Core Max') get Name, Value";
            pProcess.StartInfo.UseShellExecute = false;
            pProcess.StartInfo.RedirectStandardOutput = true;
            pProcess.StartInfo.CreateNoWindow = true;`

Little bit of string parsing and a running instance is needed, but otherwise works quite well.

elyasaf755, 2023-01-15:
I faced the same issue with my 13700k CPU. You need a higher execution level, do this:

1. Right click your project -> **Add -> New Item... -> Application Manifest File**
2. Look for `<requestedExecutionLevel level="asInvoker" uiAccess="false" />`, and change it to `<requestedExecutionLevel level="highestAvailable" uiAccess="false" />`
3. Save and restart Visual Studio

btrautmann, 2023-01-17:
Thanks @elyasaf755 I'll give this a try today and report back, but it sounds hopeful!

btrautmann, 2023-01-19:
At some point I ditched the .NET MAUI app I was using to poll the hardware sensors and instead started doing that directly from the SignalR application to make my life easier. In attempting to follow the steps @elyasaf755 laid out above, I found that I could _not_ add a manifest file to the project, but I _then_ noticed that the temperatures for the core/package were all coming through already! Not sure why the SignalR application had no problem with it, but I am inclined to believe it had higher privileges by default then the MAUI app and that everything is functioning as expected on the LibreHardwareMonitor side. I would say this can be closed if you concur @Tretgo.

[image]

closed 2023-02-05 by PhyxionNL