LibreHardwareMonitor/LibreHardwareMonitor #941: RAM Temperature

KiwiMonster364, 2022-12-25:
Hi,

I'm currently using a 12700K on a MSI Pro Z690-A DDR4 WiFi, paired with some 32GB Mushkin Redline Lumina 3600 CL16.
I'd like to poll my RAM temperature, but I can't find it. I know that my RAM has a temperature sensor, as HWiNFO64 correctly detects both DIMM temp sensors. The other information about my ram is also correctly fetched, such as utilization, etc. 
Is this feature just not available or is it an interaction error?

Keep up the good work!

KiwiMonster364, 2022-12-25:
I'm using the latest version of LibreHardwareMonitor, 0.9.1, on Windows 11, 21H2.
All needed drivers are installed. 
(I have to lookup the exact build tomorrow)

JanB97, 2024-02-02:
I would also like this. 
I am on B550 and have GSkill RAM.

JanB97, 2024-02-02:
I have tried this old commit #399 and HWMonitor HWInfo and LibreHardwareMonitor all show the same data, so I am not sure why this was not implemented.
Is there a way to implement this or a way I can help debug the problem?

[image]

Neros7, 2024-09-29:
Is there a way to help implement RAM temperatures?