LibreHardwareMonitor/LibreHardwareMonitor #948: LibreHardwareMonitor not displaying GPU temperature but Open Hardware Monitor can for the same GPU

odopo, 2023-01-04:
I have a laptop with an Intel UHD Graphics 620 and a Radeon 540 Graphics. Open Hardware Monitor shows temperature and other info like core voltages, clocks & GPU fan info for Radeon 540 Graphics that LibreHardwareMonitor doesn't show. Could you please at least add temperature info for this GPU?

[image]

[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10343327/LibreHardwareMonitor.Report.txt)
[OpenHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10343328/OpenHardwareMonitor.Report.txt)

closed 2023-03-15 by PhyxionNL

cl0rm, 2023-03-26:
Same for 6900XT (AsRock). 
OpenHardwareMonitor 0.9.2 works.
Even Report does not include the Temps

PhyxionNL, 2023-03-27:
Please attach reports from both versions, the one that works and latest LHM. I can take a look.

commit 50d7adb9 (2023-03-15): Fixes #948