LibreHardwareMonitor/LibreHardwareMonitor #950: Crashing randomly

lukes07xd, 2023-01-05:
Hi! Librehardwaremonitor crashes randomly. 
I'm using a Asus X571GD Laptop
i5 9300H
16GB ddr4 2400
512 GB NVMe SSD
gtx 1650 4gb

here's the crash report of .net framework (language of my laptop is czech):
`************** Text výjimky **************
System.OverflowException: Chyba přetečení
   v System.Drawing.Graphics.CheckErrorStatus(Int32 status)
   v System.Drawing.Graphics.DrawImage(Image image, RectangleF destRect, RectangleF srcRect, GraphicsUnit srcUnit)
   v LibreHardwareMonitor.UI.SensorGadget.DrawProgress(Graphics g, Single x, Single y, Single width, Single height, Single progress)
   v LibreHardwareMonitor.UI.SensorGadget.OnPaint(PaintEventArgs e)
   v LibreHardwareMonitor.UI.GadgetWindow.Redraw()
   v LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e)
   v System.Windows.Forms.Timer.OnTick(EventArgs e)
   v System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   v System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Načtená sestavení **************
mscorlib
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9105.0 built by: NET481REL1LAST_C
    Základ kódu (CodeBase): file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Verze sestavení: 0.9.1.0
    Verze Win32: 0.9.1.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9075.0 built by: NET481REL1LAST_C
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9065.0 built by: NET481REL1LAST_C
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Verze sestavení: 1.7.0.0
    Verze Win32: 1.7.0.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Verze sestavení: 0.9.1.0
    Verze Win32: 0.9.1.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/LibreHardwareMonitorLib.DLL
----------------------------------------
Microsoft.Win32.TaskScheduler
    Verze sestavení: 2.10.1.0
    Verze Win32: 2.10.1.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Core
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9115.0 built by: NET481REL1LAST_B
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Configuration
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Management
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
OxyPlot
    Verze sestavení: 2.1.0.0
    Verze Win32: 2.1.0.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/OxyPlot.DLL
----------------------------------------
OxyPlot.WindowsForms
    Verze sestavení: 2.1.0.0
    Verze Win32: 2.1.0.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/OxyPlot.WindowsForms.DLL
----------------------------------------
System.Configuration.Install
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll
----------------------------------------
o2hrlg3w
    Verze sestavení: 0.0.0.0
    Verze Win32: 4.8.9065.0 built by: NET481REL1LAST_C
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
HidSharp
    Verze sestavení: 2.1.0.0
    Verze Win32: 2.1.0.0
    Základ kódu (CodeBase): file:///C:/Program%20Files/LibreHardwareMonitor/HidSharp.DLL
----------------------------------------
System.resources
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.resources/v4.0_4.0.0.0_cs_b77a5c561934e089/System.resources.dll
----------------------------------------
System.Drawing.resources
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing.resources/v4.0_4.0.0.0_cs_b03f5f7f11d50a3a/System.Drawing.resources.dll
----------------------------------------
System.Windows.Forms.resources
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_cs_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------
mscorlib.resources
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_cs_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------

************** Ladění JIT **************
Aby bylo povoleno ladění JIT, musí konfigurační soubor
pro tuto aplikaci nebo počítač (machine.config) mít
v oddílu system.windows.forms nastavenou hodnotu njitDebugging.
Aplikace rovněž musí být kompilována s povoleným
laděním.

Příklad:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>`

lukes07xd, 2023-01-05:
for some reason embedding the output didn't work as expected so it looks weird...

PhyxionNL, 2025-10-01:
Dupe of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/423.

closed 2025-10-01 by PhyxionNL