LibreHardwareMonitor/LibreHardwareMonitor #954: RX 7900 XTX GPU Package always 0.0W

Kyle2142, 2023-01-11:
[image]
All other sensors seem fine. I would likely be able to assist finding the right sensor via a debug build or something - Not sure how you'd find it without having this card yourself

jonvel, 2023-04-25:
Sigh, I hate to "me too!" this, but ... yeah.  Me too.  Though note that HWMonitor64 is able to read the sensor (so it must exist somewhere), though OpenHardwareMonitor is not (sensor doesn't even show up).

NecromancerForLife, 2023-08-31:
fixed with topic https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1074

Kyle2142, 2023-09-02:
Hi. Thanks to those involved for their time.
using this (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/6048599802) master build artifact, I have a very ... interesting result.
[image]
Steps to reproduce:
1. Open up a game (or relatively constant GPU load program), LHM and AMD Adrenaline. Latter two preferably on a 2nd monitor
2. If a game: Keep the scene still for stable power draw statistics on Adrenaline
3. Add `GPU Package` to Plot and observe LHM side by side with Adrenaline

Observed:
- The power draw in LHM is not equal to Adrenaline. Sometimes close, sometimes far off
- The power draw plotted is not equal to the table value. As pictured it seems to be wildly off by -50 or so watts

The latter is more concerning than the former. Why would the table and plot differ at all?
Concerning the former, I understand AMD has not made public the method/calculation.
While it would not be reasonable to ask LHM's contributors to "make this perfect" it also is not reasonable to rely on the data as given.
BUT I have similarly weird experience with the graphs in Adrenaline. They seem to re-calculate the graph in some weird way which causes inaccuracies. While I cannot say this affects TBP or not, I have mostly observed it in the frame time. I doubt it is relevant so I can elaborate / record an example if asked

NecromancerForLife, 2023-09-02:
> Hi. Thanks to those involved for their time. using this (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/6048599802) master build artifact, I have a very ... interesting [...]

Check this page: this is the Total Graphics power now not the Total board Power. https://www.igorslab.de/amd-radeon-rx-7900xtx-und-rx-7900xt-im-test-ein-riesiger-schritt-nach-vorn-und-ein-kleiner-zur-seite/11/

I also saw this peaky behavior with my 7900 XT in the other Thread 1usmus said this would be totally normal and explains why you need 750W Power supply minimum.

Kyle2142, 2023-09-02:
Maybe I don't understand something but it is very obvious from my image that LHM's table and plotter do not match. It is clear that the maximum in the plotter in the last few datapoints is around 80W which is far from both the table (167W) and Adrenaline (140W)

NecromancerForLife, 2023-09-02:
> Maybe I don't understand something but it is very obvious from my image that LHM's table and plotter do not match. It is clear that the maximum in the plotter in the last few datapoints is around 80-90W which is far from both the table (167W) and Adrenaline (140W)

no i misunderstood. i thought it was about the difference in lhm and amd radeon. I do not know how this graph works.

closed 2026-01-03 by PhyxionNL

jonvel, 2026-01-11:
Interestingly, the "Power (W)" _plot_ appears to be exactly 1/2 of what the Widget (and the main interface), and the Adrenaline software, shows for Total Board Power line for my 7900 xtx.  It seems to track very closely.