LibreHardwareMonitor/LibreHardwareMonitor #966: Can't Access Temperature Data on Intel Core i9-12900H

kzettlmeier, 2023-02-06:
Looks like the values for Temperatures are blank for the CPU.  You can find the report attached.
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10669033/LibreHardwareMonitor.Report.txt)

kzettlmeier, 2023-02-06:
[image]

atomo33, 2023-02-09:
In my case Avast said LibreHardwareMonitorLib.sys (LibreHardwareMonitorLib.dll) is as an unsafe driver and disabled it. Thus no more Temperature monitoring for me.

Drew221, 2023-02-10:
Yea Avast One blocked it and the old Open Hardware Monitor for me too. Not sure how to get around it.

atomo33, 2023-02-10:
In my case, (again), I disable the new Avast driver protection, then I run LibreHardwareMonitor, and then I enable the Avast protection again.

PhyxionNL, 2023-03-15:
#984

kzettlmeier, 2023-03-15:
To be clear, I am not using Avast.

PhyxionNL, 2023-03-15:
Ah, but it's related to Ring0, from your report:

Status: Installing driver "LibreHardwareMonitor.sys" failed and file exists

Perhaps you're running something else that uses WinRing0?

kzettlmeier, 2023-04-07:
@PhyxionNL everything seems to work fine in that build now!  Thanks and sorry for the late response!

closed 2023-04-07 by kzettlmeier