LibreHardwareMonitor/LibreHardwareMonitor #971: In config, switching on GPU Memory display causes Bus to be shown as well.

hdf, 2023-02-12:
The problem originates from here:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/62ffe8922e45dd0a33a8b4d8ebe3f52e744ad770/LibreHardwareMonitorLib/Hardware/Gpu/NvidiaGpu.cs#L424
There is an index collision. Bus already has the index of 3.

hdf, 2023-02-12:
Setting the index from 3 to -1 seems to work without any problems in this case, and seems to be the easiest solution.

ntoff, 2023-02-18:
I don't think setting it to -1 is the right way. 

I think I found a more dynamic way, that *should* work no matter how many load indexes there are.

Delete line 424 and move it to line 401 (under `_gpuSharedMemoryUsage`) with the contents:  `_memoryLoad = new Sensor("GPU Memory Load", nodeSensorIndex++, SensorType.Load, this, settings);`

That should auto increment the node index, and then let the other auto-incremented indexes do their thing.

hdf, 2023-02-18:
@ntoff Agreed, that is a better solution and seems to produce no problems. Thank you!

commit ad575e7a (2023-03-15): Fixes #971

Closes #972
closed 2023-03-15 by PhyxionNL