LibreHardwareMonitor/LibreHardwareMonitor #984: WinRing0 get's blocked by Avast/AVs

Rem0o, 2023-02-23:
There's a CVE for WinRing0 https://www.cvedetails.com/cve/CVE-2020-14979/
See also https://posts.specterops.io/cve-2020-14979-local-privilege-escalation-in-evga-precisionx1-cf63c6b95896

@PhyxionNL  What do you think of the proposed changed in the driver 
(replacing https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iocreatedevice to https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdmsec/nf-wdmsec-wdmlibiocreatedevicesecure) to remove the potential privilege escalation exploit?

PhyxionNL, 2023-02-23:
I'm aware - I have no problem with updating WinRing0, it would even make it possible to drop InpOut by building it with PHYSICAL_MEMORY_SUPPORT set. But the driver needs to be signed, which requires an EV Code Signing certificate. The driver also needs to pass HLK checks as kernel drivers needs to be signed by MS.

Rem0o, 2023-02-23:
Is the current sys file here signed? Who built it?

PhyxionNL, 2023-02-24:
Yes, it's signed (by Noriyuki MIYAZAKI from CrystalMark), otherwise it wouldn't have worked. I'm not sure it was signed by an EV certificate either, this wasn't mandatory in the past and Windows allows these older signed drivers to work in order not to break them, but new drivers first need to be signed by EV so you can upload them to MS' hardware dev center, which then signs the driver with their own certificate. New kernel mode drivers need to be signed by MS otherwise Windows rejects them.

Rem0o, 2023-02-25:
Will do some testing with the WdmlibIoCreateDeviceSecure. I will compile the kernel driver with it, then disable secure drivers on my dev machine and see if functionalities and everything are maintained.  If that works, might try to get the whole loop of EV signing and everything going. I don't mind paying for it.

Rem0o, 2023-02-25:
Update:
[image]

Applied the changes suggested in the article with the "secure" version of the device object.  Compiled that, put it in LHM, rebooted into no-sign kernel driver mode, and it just works.

PhyxionNL, 2023-02-25:
Nice work 👍 Please set _PHYSICAL_MEMORY_SUPPORT then as well so we can ditch InpOut too. BTW, the driver version in the repository here (1.2.0.5) isn't the latest version (1.3.0.x). WinRing0 isn't downloadable anymore on the openlibsys website, but there are some GitHub projects that have a newer version: https://github.com/QCute/WinRing0/releases, and this https://github.com/GermanAizek/WinRing0 although the project files are GPL so you might need to create a new project yourself (all original files still seem to be under the original BSD license). I'm attaching a 1.3.0 version I found on the web that seems legit (mirrored originally from sourceforge).

[WinRing0_1_3_0.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10832036/WinRing0_1_3_0.zip)

Rem0o, 2023-02-25:
Checked the sources for the zip you sent, and it still seems like 1.2.0.5
[image]

By the way for the _PHYSICAL_MEMORY_SUPPORT thing, you'll have to be more precise.  I'm really not familiar with Windows kernel stuff.  I will probably open a repo for this.

PhyxionNL, 2023-02-25:
Ah, maybe it's the same version then, a bit weird that that's not increased with 1.3.0 🤔 Might be worth comparing the files.
_PHYSICAL_MEMORY_SUPPORT  is a preprocessor directive. You can add it VS project settings, or modify the source a bit to use the code between the ifdef.

I think you indeed need to be an organization for EV, and probably have that organization registered somewhere. I don't think a "GitHub organization" will suffice. You shouldn't have to renew if you don't need to resign the driver (it'll stay valid if you timestamp it and MS should do that for you). From a legal point of view the BSD license should take care of that ([wiki](https://en.wikipedia.org/wiki/BSD_licenses#3-clause_license_(%22BSD_License_2.0%22,_%22Revised_BSD_License%22,_%22New_BSD_License%22,_or_%22Modified_BSD_License%22))).

CrazyKidJack, 2023-03-14:
> Also also, looked at the Extended validation Code signing stuff, and it seems like you have to be registered as an organization to be able to get the cert and renew it year after year. Is there any example of open source kernel drivers being used at large? If said association is myself, any usage of that driver will fall under my responsibility and it might get ugly with the legal stuff. Also, I'm no organization, just a solo dev.

@Rem0o 
Not sure where you live but where I live I think I could start a sole proprietorship for free and get an EV cert that way. While starting the "company" would be free it is a bit time consuming so if there are other options, I'd rather those be pursued first... but if that is the only option, I'd be willing to do it if we could work something out for the EV cert.

I know I'm a rando... but if you are interested in chatting about it, you can send me an email here: fancontrolsevcert@gmail.com

Rem0o, 2023-03-14:
@CrazyKidJack  I don't think that would work either. Goal of EV signing is to avoid bogus entities.  There is a whole process of verification, with address and everything.

CrazyKidJack, 2023-03-14:
> @CrazyKidJack I don't think that would work either. Goal of EV signing is to avoid bogus entities. There is a whole process of verification, with address and everything.

I aware, and it would not be bogus. It would be a real sole proprietorship / real company. That company just would not produce anything, would have have any expenses, and would not make any revenue.

SearchForTheCode, 2023-05-09:
So there is a solution to the CVE but i cannot be applied, because the driver must be signed properly, is that correct? :( thats sad.

SearchForTheCode, 2023-05-11:
> BSD license

Did you set

> Update: [image]
> 
> Applied the changes suggested in the article with the "secure" version of the device object. Compiled that, put it in LHM, rebooted into no-sign kernel driver mode, and it just works.

Did you set up a repo with those changes? Would be nice to test them for myself :)

Rem0o, 2023-05-11:
@SearchForTheCode 
code is here https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/tree/master/WinRing0

Just modified that 1 line and compiled.

SearchForTheCode, 2023-05-12:
Ah nice, did the same , it seems to work but suddenly did not read any CPU temperature anymore. Need to dig into it. 
But i can confirm that Avast Business is not blocking it anymore with that changes.

mtaalas, 2023-06-05:
It seems that this is the relevant to why this is happening?
https://nvd.nist.gov/vuln/detail/CVE-2020-14979

This has broken many other projects as well. But yeah, security is ever changing field and software just needs to keep up...

Thank you for your hard work everyone. You're awesome!

namazso, 2023-06-10:
I have a (signed) driver that could be used as a replacement, however it'd require quite some changes to the code. Plus, I have the signing keys for the security features of the driver that prevent it being a WinRing0 v2, but that'd mean I'd need to review and sign all the changes affecting code that needs kernelmode features.

I'm not sure how comfortable maintainers here would be with that, so I might just end up with a fork.

namazso, 2023-06-10:
If you're wondering about why signing and "why not just admin restrict the device", that's because anticheats would still hate that. Plus it'd go against multiple items on [Microsoft's recommendations](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#do-not-production-sign-test-code):

> ## Do not production sign test code
> 
> **Security checklist item \#4:** *Do not production code sign development, testing, and manufacturing kernel driver code.*
> 
> Kernel driver code that is used for development, testing, or manufacturing might include dangerous capabilities that pose a security risk.  This dangerous code should never be signed with a certificate that is trusted by Windows.  The correct mechanism for executing dangerous driver code is to disable UEFI Secure Boot, enable the BCD "TESTSIGNING", and sign the development, test, and manufacturing code using an untrusted certificate (for example, one generated by makecert.exe).
> 
> Code signed by a trusted Software Publishers Certificate (SPC) or Windows Hardware Quality Labs (WHQL) signature must not facilitate bypass of Windows code integrity and security technologies.  Before code is signed by a trusted SPC or WHQL signature, first ensure it complies with guidance from [Creating Reliable Kernel-Mode Drivers](../kernel/creating-reliable-kernel-mode-drivers.md). In addition the code must not contain any dangerous behaviors, described below.  For more information about driver signing, see [Release driver signing](#execute-proper-release-driver-signing) later in this article.
> 
> Examples of dangerous behavior include the following:
> 
> - Providing the ability to map arbitrary kernel, physical, or device memory to user mode.
> - Providing the ability to read or write arbitrary kernel, physical or device memory, including Port input/output (I/O).
> - Providing access to storage that bypasses Windows access control.
> - Providing the ability to modify hardware or firmware that the driver was not designed to manage.

javs, 2023-06-10:
This would be great IMO! Would your new driver be open source?

Also, if you are not planning on exposing arbitrary memory mappings, would that mean we would move some of the logic that is today in LHM to the kernel driver?

namazso, 2023-06-10:
> Would your new driver be open source?

Possibly, I haven't decided yet. The other option I'm considering is source-available.

> Also, if you are not planning on exposing arbitrary memory mappings, would that mean we would move some of the logic that is today in LHM to the kernel driver?

It implements an interpreter that you can feed signed programs into.

mtaalas, 2023-06-10:
> [...] some restrictions to make sure commercial entities don't just straight up copy it (and just sell alternative licensing to them).

This... 100% this. There are licenses that prevent commercial use of source code and/or if any part of that code is used, the resulting program _must_ be open source as well.

And it's really important to choose the right license. You should talk with OSS people about the best way to ensure your work won't benefit commercial entities unless they're ready to open up their systems as well.

Then, if they do break the license, you can sue them and as billions for damages :D

Without proper license, you should never open up open source code that you want to keep free and open.

I'm so angry about the OBS thing... OSS community definitely learned a lesson there.

namazso, 2023-06-10:
> There are licenses that prevent commercial use of source code

That's not open-source, that's source available - open source requires that there are no restrictions on field of use.

> and/or if any part of that code is used, the resulting program must be open source as well.

Microsoft also has rules about what they sign. GPLv3 has clauses about giving up keys if signature enforcement is present (and therefore a no-go for Microsoft), so if I just licensed it as that, it'd be as useful as giving no rights at all, since OSS projects couldn't use it for its single purpose.

In the meantime, open-sourcing their fork might not be a big deal for these companies either, as all they'd want to change is the device path and the public key burned into the driver. So I wouldn't see a penny either from their very commercial usage.

So yeah, lots of considerations, which is why I didn't just dump it on GitHub already. But let's just stick to technology in this issue. It will be out in some shape or form, and certainly free as in beer to non-commercial users.

PS. I'm also available on mail and a bunch of chat platforms for more direct chat.

namazso, 2023-06-10:
Decided to [publish the source](https://github.com/namazso/PawnIO), for now without any extra rights given to anyone until I figure that stuff out.

namazso, 2023-06-11:
Sorry for spamming this issue, but I'm just interested in any feedback.

I updated the license of [PawnIO](https://github.com/namazso/PawnIO) and released an example module in a new [modules repo](https://github.com/namazso/PawnIO.Modules) that should give an idea how they will look. This was already tested on a live version of the driver. The natives are not documented for now, nor is basically anything else, this is something I plan to address in the future.

Some tooling and libraries will be the next thing I'll release, followed by binary releases. These will most likely be proprietary (for some control over distribution format, enforcing uniform installations to avoid what happened to WinRing0), but otherwise identical to the released code.

It will come in 2 flavors:

- Unrestricted (accepting any blobs), test signed.
- Restricted (signed by me only), Microsoft signed.

The first can be used for development and testing, while the second for public releases.

javs, 2023-06-13:
I'm interested because it would kill the two kernel drivers in here.

I would be willing to help migrate  if we go there.

You probably need to have @PhyxionNL check this out, seems he's currently a bit backed up by looking at the open PRs.

SearchForTheCode, 2023-06-13:
Wouldnt that change the overall license of Libre with currently using MPL 2.0 ?

namazso, 2023-06-13:
Not exactly. MPL is weak copyleft, and therefore allows for linking against proprietary things (ie my binary distribution). The source distributions I posted (Modules: LGPL 2.1, Driver: GPL 2 + interface exception, Library: LGPL 2.1) all also not viral on this interface, so even if I disappeared and someone had to fork those from me, that still wouldn't affect licences here. The only license that would be (indirectly) affected is the actual hardware interfacing code that'd have to be rewritten into PawnIO modules, and that must always be a GPLv2+exception compatible license.

That means if you were to fork LHM and transform it into a different product (to allow closing source as per MPL), then forked and signed a different version of my driver, you'd still need to open up the PawnIO modules. Unlike now, where you can extend your private fork of LHM with the extra hardware you probably manufacture.

But this is intended on my side, I want hardware manufacturers to either open up the comms (so that their hardware can be added here), or contact me for a different (paid) license on PawnIO's source code.

I think this is an acceptable compromise.

SearchForTheCode, 2023-06-13:
Absolutly if there is a paid license this is always a good comp and very fair!! But if you just use this library as it is via the nuget package, the MPL-2.0 will be the main license because Libre has published their source to PawnIO Modules, correct? ( i hate those complicated license stuff, dont hate the licenses itself :D)

namazso, 2023-06-13:
Good question.

How inlined blobs of data affect licenses is up for debate (see Linux kernel firmware blobs) and so you might be able to just inline the LGPL blobs and still claim MPL. It is commonly believed that GPL stretches as far as the address space goes. Pawn modules are ran in a VM, so the "infection" should not be an issue.

As for distributing the driver itself, I was hoping for a model similar to Wireshark and NPcap. Basically you have a separate installer that just installs NPcap, and then everything can use it.

PhyxionNL, 2023-06-13:
I'm not opposed to using a different driver, but the license needs to be compatible with the project. Even with the exception clause it would block usage for a lot of programs currently integrating it.

namazso, 2023-06-13:
>  If we were to include your driver then LHM would pretty much be GPL licensed.

That certainly isn't the intention! The exception on the GPLv2 is there for [PawnIOLib](https://github.com/namazso/PawnIOLib) which this way can be LGPL, which means anyone can link arbitrarily licensed LHM forks to it.

The idea is that you'd be distributing the LHM code under MPL, and ask users to install PawnIO, which is just the lib and the sys. This **would** be a slight change in behavior, however shouldn't affect licensing affairs at least.

Why do you think the exception is insufficient?

namazso, 2023-06-13:
> LGPL is not 100% compatible with MPL 2.0 afaik

There are some nuances when combining actual code, but both licenses allow dynamic linking to arbitrarily licensed code, including fully proprietary.

> That would largely address LHM code licensing but not its usage.

Can you expand on this? I'm sure we can work out something for whatever use case there is out there that wouldn't work.

> It's also not very convenient for portable users that they have to install PawnIO separately.

I'd rather see it as how Wireshark portable works - you can look at packet captures, but if you want to make a local capture you need to install NPcap. Same could be here except on a per-device basis.

> One integrated package that transparently installs it would be much easier for end users (similar to WinRing0).

I'm not so sure. There's a [constant](https://github.com/nicehash/NiceHashMiner/issues/2537) [stream](https://www.reddit.com/r/raptoreum/comments/r6uxch/winring0_and_msr_issue_help/) [of](https://github.com/xmrig/xmrig/issues/2683) [issues](https://www.reddit.com/r/razer/comments/r82j2j/razer_synapse_new_conflict_with_winring0_driver/) where multiple programs ship the same WinRing0 while registering it under different names and places. I'd like to fix that.

namazso, 2023-06-13:
> LHM only has a portable edition, and FanControl is also just a portable edition.

Well, I'd argue about whether registering a system-wide service is a "portable" thing to do. Running kernel drivers from user-writable directories is also a bad idea in general, it should be done from an admin-only dir (like program files).

If you want to do it right, you'll end up with what's commonly called an installer. Just silent at most.

> I can already see tons of reports about missing sensors. You'd basically have to add a big message that a separate driver install is required.

I was thinking more like a warning popup on startup like "Blah blah, PawnIO is required for many sensors. Start installer?" that should reduce user friction. Especially if the installer is bundled.

For LHMLib it can just be marked as a "breaking change". Considering a few bits would need to be rewritten, that's a pretty good idea in general too. It also would be a breaking change anyway, as people who had WinRing0 installed by other software would need to run it as admin now, while the with the vulnerable version it just worked.

---

The only solution other than an installer I can see is some convoluted "is another copy of me already running" on startup and "is someone else using my copy" and "am I the last one using this someone else's copy" on shutdown logic. Plus copying to a system directory on start and deleting when needed. Which is very overcomplicated and error-prone.

namazso, 2023-06-13:
> It would then be useful to include a function in PawnIOLib to check if the driver is installed and which version is installed so we can take action on that.

Indeed. The idea was to just test for PawnIOLib's presence.

> Would PawnIO also allow us to get rid of InpOut? I'd be nice to only use one driver.

Yes. These are the primitives it provides to pawn modules: https://github.com/namazso/PawnIO.Modules/blob/main/include/native.inc

Now, since these are **very** powerful, I obviously won't sign rootkit-tier code like winring0, but adding the bare minimum of code for ensuring you're talking to the right device and doing the talking should be fairly easy. Tests can be done with the unrestricted driver, and signed blobs can be used with the properly signed one.

> A PR with PawnIO integration would be appreciated, it'd probably be easier for you 🙂.

Well, moving the reading into pawn module(s) would take a community effort, but I can do a single device easily. Since I have a Ryzen CPU, might just go with porting that.

> Oh, your PawnIO.Releases repo isn't available (or private).

Yes, I've been slowly publishing the various components as I got them cleaned up and ready. I still need to add some finishing touches to that one.

PhyxionNL, 2023-06-13:
I think those drivers ended up there because they're actually vulnerable, not because of the ability to read and write artibrary data. 

I'm not sure how useful PawnIO would be otherwise, as it would:
- Probably require quite some modules.
- Rely on you to build and sign each module. That's OK if this is a one-time thing, but if a new device/component needs to be added, it probably wouldn't work with the existing module, so we'd back to square one.
- Probably be too difficult for the community to implement all these different modules to get the same functionality as-is.

namazso, 2023-06-13:
> I think those drivers ended up there because they're actually vulnerable, not because of the ability to read and write artibrary data.

That's varying. The Admin-only RTCore64 used to be there for like half a year. It was removed recently, but it might be put back in the future. Basically no one really likes these kind of drivers because often they're often abused for admin -> kernel. Even though that's *supposedly* not a security boundary (although Microsoft always patches such exploits in their own code), malware frequently ships and uses them, so they have awful reputation.

> I'm not sure how useful PawnIO would be otherwise

Well, that is unfortunate, but understandable. I might just go ahead with a fork instead then (trying to make this not sound like a threat, we just have different incentives, like you making sure vital dependencies don't die, and me trying not to get my driver blacklisted).

I also didn't like certain architectural choices here anyways (like requiring admin, instead of splitting into service (admin) + client (user)).

> Probably be too difficult for the community to implement all these different modules to get the same functionality as-is.

Most modules would be a series of like 3 io reads and stuffing it into the out array. It takes a bit of time, but certainly not skill. I guess OSS projects don't exactly have so much of that either.

Rem0o, 2023-06-13:
> Well, that is unfortunate, but understandable. I might just go ahead with a fork instead then

@namazso  I followed the conversation along seeing where it would end up.  The alternative proposed could be promising, but in any case I think we would need to see a working prototype (in a branch/fork) anyway with everything working to see if it looks viable to go ahead and port everything away from WinRing0 in the master branch here.   I'm curious as to exactly how it would look, as if I understand correctly, every "arbitrary" memory mapping written right now in Code as Ring0.Read( address ) / Ring0.Write( address ) would need to be hardcoded in a small module, which would then need to be signed, and packaged/integrated into the app. 

> I also didn't like certain architectural choices here anyways (like requiring admin, instead of splitting into service (admin) + client (user)).

That's just for the winform LHM app itself, which I see as a functional/useful demo app for LHMLib. It just so happens to be a simple 1 process application that is "portable". I can very much see a LHMService folder being added here in the future, for any client to be built on top with gRPC/REST API/IPC or whatever.  I don't see how the LHM client/app is relevant to this conversation.

PhyxionNL, 2023-06-13:
> Most modules would be a series of like 3 io reads and stuffing it into the out array. It takes a bit of time, but certainly not skill. I guess OSS projects don't exactly have so much of that either.

It's not just writing the modules, it's also relying on you to sign and build them. That's doesn't have to be a huge problem if the modules are flexible enough to not require that for every change. Take this recent commit for example, https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/701e8ad5143d6d249006429cfd12511ccd01d6c6, this added some ReadByte's for the Ring0 driver. This is the kind of flexibility that should be possible to achieve without rewriting the modules. Most of the PRs that add new sensors are from users that have a system with a sensor and they need to be able to try their changes out. If every change they make requires a rewritten module, it's not feasible for either party.

What about some other ways to secure the driver? Such as checking if the calling process has a trusted certificate. Most AVs/FW also show that data, such as:
[image]

That all said, I'm more than happy to take a look at a PR and see what's what. I don't know what you have exactly in mind, but if we'd end up keeping WinRing0 in addition to the PawnIO driver then we just have yet another driver.

> I don't see how the LHM client/app is relevant to this conversation.

Exactly.

namazso, 2023-06-13:
Btw, about the two driver thing - you could use a better (but still vulnerable) driver to get rid of one:

https://crystalmark.info/download/archive/CrystalCPUID/CrystalCPUID415Src.zip
https://crystalmark.info/download/archive/CrystalCPUID/CrystalCPUID415x64.zip

namazso, 2023-06-13:
@Rem0o 

>  The alternative proposed could be promising, but in any case I think we would need to see a working prototype (in a branch/fork) 

I'll probably work on that in the following weeks.

> every "arbitrary" memory mapping written right now in Code as Ring0.Read( address ) / Ring0.Write( address ) would need to be hardcoded in a small module, which would then need to be signed, and packaged/integrated into the app.

That is correct.

> I don't see how the LHM client/app is relevant to this conversation.

Just mentioned another thing I wanted to change if I went forward with a fork, not that relevant.

@PhyxionNL 

> users that have a system with a sensor and they need to be able to try their changes out.

That's perfectly doable with the test signed version - it requires no signatures on the modules.

> Such as checking if the calling process has a trusted certificate.

Unfortunately that's largely useless (and incorrect in the AV). Another app can easily just start a suspended copy and inject. Multiple HW vendors implemented such checks, and they're all frequently circumvented.

> That all said, I'm more than happy to take a look at a PR and see what's what.

Sure thing. I could also do an usermode reduced PawnIO implementation on top of WinRing0, in case that could help anything (like upstreaming the changes but keeping WinRing0 / keeping it optionally)

namazso, 2023-06-13:
I see, OK. I'll look into actually porting things, so that I can have a better sense of just how complicated all of it would be.

namazso, 2023-06-14:
Well, I ["ported"](https://github.com/namazso/PawnIO.Modules/commit/00ffa7d00a3f39edba0eaa27361752d115f279b4) Ryzen SMU and it indeed looks more like a rewrite at this point. It was slightly more challenging because my SMU is not actually supported by LHM (Raphael, 0x540104) and I just added some made up size (0x1000) when testing, but just the porting part itself took multiple hours.

I still need to integrate it in LHM, since so far I just compared the floats it read to make sure it works at least.

PhyxionNL, 2023-06-15:
Sounds good. It might be a good idea to also separate the read/write ranges as more ranges are being read than written. Obviously some writes still need to happen as it would otherwise be impossible to control fans 🙂

alberts8, 2023-06-16:
@namazso
I guess that msr and mmio register reading and writing on intel cpu's would be doable that way as well. Would it be necessary to limit which registers can be used or would it be enough to restrict such a module to only have access to the msr's, pci configuration and the whole mmio register range.

namazso, 2023-06-16:
Tough question, I'll see when I get there, it really depends on what we might accidentally hit with a range. For example for MSRs reading can at most leak KASLR addresses, but arbitrary write is enough for complete kernel takeover. There's multiple ways even, you can overwrite KERNEL_GS_BASE to run on user stack, or overwrite LSTAR so it points past the swapgs...

In the meantime, I got something done with LpcIO: https://github.com/namazso/PawnIO.Modules/blob/main/LpcIO.p

It didn't seem too dangerous to just give full access to the Super I/O, but I did move the enter/exit and detection logic inside so that I can handle Gigabyte EC later.

Seems to work for my Nuvotron NCT6686D as far as detection goes. I should probably spend some time with integration and getting some distribution package of PawnIO going, but I might as well run into [all the bugs](https://github.com/namazso/PawnIO.Modules/commit/d6cb70d6b0778dcbd6577b91b6e7c6bfe6e6e373) in my driver before a public release.

SearchForTheCode, 2023-06-21:
> Update: [image]
> 
> Applied the changes suggested in the article with the "secure" version of the device object. Compiled that, put it in LHM, rebooted into no-sign kernel driver mode, and it just works.

I also tried this code for testing purposes. And while all CPU related stuff seem to work, i got problems on the mainboard parts. Voltages and Mainboard Temperatures are not detected any more as sub hardware. I only applied those changes and booted in the non-signed kernel driver mode. CPU Temp works like normal. 

Any explanation why this could be?

chmichael, 2024-09-20:
Here's the latest WinRing0 repository: https://github.com/GermanAizek/WinRing0
Someone needs to compile and and sign it.

PCAssistSoftware, 2025-02-26:
Hi, just a quick message to find out if any progress or plans on this relating to WinRing0?  I love this project but can't use it on so many systems due to the blocking of WinRing0 and wondered what (if anything) we as a group could do to resolve it?

ColbyBouma, 2025-03-11:
Windows Defender just started blocking this on my personal computer (Windows 11 23H2). It took a while to track down because it was showing up as `C:\Windows\System32\WindowsPowerShell\v1.0\powershell.sys`. I don't know if that's an artifact of calling `LibreHardwareMonitorLib.dll` from my CheckMK plugin. This was really annoying because Defender kept popping up over and over 😄 

I tried updating to v0.9.4, but that version of `LibreHardwareMonitorLib.sys` (WinRing0.sys) is [the same](https://www.virustotal.com/gui/file/11bd2c9f9e2397c9a16e0990e4ed2cf0679498fe0fd418a3dfdac60b5c160ee5). I was hoping it might have been updated without this issue getting updated.

That SYS file didn't exist until I called the DLL for the first time. I've been running LHM as a CheckMK plugin for nearly 2 years, and this problem just started with the latest Defender definition update, Version 1.423.334.0.

namazso, 2025-03-12:
Well, last time when I offered my driver I got a bit bored of porting devices I don't even own over...

Anyway, I pushed out a repository with a signed and unsigned version:

https://github.com/namazso/PawnIO.Setup

Unsigned one can load unsigned modules, while signed only loads signed modules. I ported a couple hardware things over here last time:

https://github.com/namazso/PawnIO.Modules

These will be signed for use with the proper driver once I set up some CI pipeline.

This commit ports Lpc functionality in LHM (at least in the version 2 years ago) over to PawnIO:

https://github.com/namazso/LibreHardwareMonitor/commit/69d9ad8f85e09036be98a443cd1bd15dab02618e

The module in there is also signed and should work with the production driver.

I don't know if there's still any interest in moving over, but it is at the very least relevant again.

irusanov, 2025-03-12:
I've took a look in the EVGA driver and it is a mix between WinRing0 and Opcode from openhardwaremonitor.

Btw, maybe worth mentioning that the way LibreHardwareMonitor handles the driver kind of minimizes the risk. It's not ideal, but still better than nothing.

The driver and service are installed on each launch and only active while the app is running. On launch the driver is extracted and the service is created. On each close the service is removed (unless it fails to stop it). Furthermore the service is additionally restricted to System and Built-in administrators with this code:

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/b3f66214e0abcc1efa0eac465434122e65840477/LibreHardwareMonitorLib/Hardware/KernelDriver.cs#L83

It also does not install WinRing0x64.sys in System32 directrory as the original DLL does.

namazso, 2025-03-12:
I’m not so sure the EVGA driver won’t get banned: Microsoft blocked Process Hacker’s driver for being able to kill PPL processes. All physmem drivers can do that via DKOM, so it might be just a matter of time until some malware uses it like how it happened to PH.

That’s why I wanted to avoid developing such a driver myself. Though ofc it’d be a really cheap and quick fix to the current situation.

irusanov, 2025-03-12:
So I got another update "2025-03 Cumulative Update for Windows 10 Version 22H2 for x64-based Systems (KB5053606)" and now it's not blocked anymore.

namazso, 2025-03-12:
@irusanov The reduced functionality one is just as powerful, with MSR writes you can overwrite kernel GS, and make the kernel run on user stack. That can be exploited for ROP and ultimately full kernel code execution.

namazso, 2025-03-13:
> I'm trying your Libre branch. Installed the "unrestricted" PawnIO release, and I was able to run the included module without disabling driver signing. Is that expected?

No, not at all. I checked, accidentally the unrestricted installer also packs the restricted and signed driver, oops! I'll fix that in a bit. Your computer should forbid even loading the testsigned PawnIO, so nothing would ever work, regardless of whether the pawn modules are signed or not.

> Unsigned module:

That one is actually signed. The windows properties only shows authenticode signatures, you can instead look with a hex editor. The structure of PawnIO modules is the following:

```
u32 signatureSize;
u8 signature[signatureSize];
u8 pawnModule[];
```

> Also, it is mentionned the signature is provided by https://reaps.gg/. What's that entity?

At the time this was signed, it was an in-development software licensing (DRM) platform of a friend (hence the driver signing cert), nowadays just a generic software contracting business.

vintagesucks, 2025-03-13:
https://www.theverge.com/report/629259/winring0-windows-defender-fan-control-pc-monitoring-alert-quarantine [[via](https://github.com/Rem0o/FanControl.Releases/issues/3016#issuecomment-2722642260)]:

> But there may be a simpler alternative: why not fix the vulnerability in WinRing0 itself? To my surprise, three developers tell me that WinRing0 has [already been patched](https://github.com/GermanAizek/WinRing0/issues/9), but the open source community doesn’t believe they can afford to get a new version signed by Microsoft — and without Microsoft’s digital signature, Windows won’t let users install it to begin with.

> But there is some hope. Prebuilt gaming PC manufacturer iBuyPower, whose Hyte Nexus monitoring software also uses WinRing0 and got flagged by Windows Defender, tells The Verge it will endeavor to get an updated WinRing0 signed — and give the results back to developers.
>
> “If this solution works, we’ll share our updated and signed version of the library, so the community of developers can distribute new versions of their apps with validated Microsoft drivers,” Hyte product director Robert Teller tells us.

Rem0o, 2025-03-15:
Tested an intel module today with PawnIO in LHM:
https://github.com/namazso/PawnIO.Modules/issues/1#issuecomment-2726768393

This could work. 

Also, OCBASE made an announcement that they want to release a driver alternative for the community with a fairly permissive licensing model:

So far for alternatives I've seen:
* IBuyPower through Hyte make a claim they want to release a "WinRing0 2"
*  OCBase [to release a community first kernel driver](https://www.youtube.com/watch?v=xPGkoZU2rvY&t=2s&ab_channel=OCCT-OCBASE)
* @namazso with PawnIO which prototypes do work. Dev workflow still needs some work though.

namazso, 2025-03-15:
Obviously I'm biased regarding which option I support, however I'd like to say that whichever ends up happening, it really should support a scriptable interface. Current LHM contains some [questionable code](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Cpu/Amd10Cpu.cs#L114-L129) that can end up leaving your CPU underclocked if LHM closes / crashes at the wrong time. The fix is to disable interrupts while doing that, but no user-mode code must run with interrupts disabled, so there's no way to fix this other than either moving this whole thing into the driver, or having a driver that you can script. I think the later of those options is the future-proof one.

PhyxionNL, 2025-03-17:
My issues with PawnIO:
- We'd have to rely on you (or someone else) to sign the thing. Given that changes to a module means it needs to be resigned, that's not really future-proof and will be problematic if we cannot automate this.
- It's not as developer friendly as being able to write it like we can right now.
- The driver is GPL, and despite some exception clauses, this would be incompatible with some of the software using LHM.

The other two drivers, while not ready, sound like a good solution. Let's hope this'll be released before MS blocks it again.

Rem0o, 2025-03-17:
@PhyxionNL  
Keep in mind both the OCBase and IBuyPower drivers would have to have some kind of constraints on the available addresses. Not much different than what @namazso is proposing here.   I don't think we can expect a "signed once" all you can eat kernel driver going forward, that's just not possible with Microsoft constraints.   I'm in contact with the OCBASE guy to see what he's proposing. I made the whole argument for the community dev workflow to be as seamless as possible. One advantage is they too need pretty much the same thing as LHM needs with OCCT, so some work will probably be done and signed by them ahead of time when new hardware gets released.

> Given that changes to a module means it needs to be resigned, that's not really future-proof and will be problematic if we cannot automate this.

What do you expect as far as signing goes? I don't see how any actor offering kernel driver signing will automatically/"blindly" accept any change to the driver.

> It's not as developer friendly as being able to write it like we can right now.

The difference is you will have to dev in unrestricted mode for testing, then submit.  With "smart" ranges set, hopefully that won't happen too often.

[28 more replies]

PR #1857 (2025-09-16): Swap WinRing0 to PawnIO
Fixes #984

TODO:

- UX choices
- Error handling (?)
- Testing

Known regressions:

- Gigabyte EC is unimplemented for now
closed 2025-09-16 by PhyxionNL