LibreHardwareMonitor/LibreHardwareMonitor #985: Replacement (Newly manufactured?) NZXT Kraken x63 not detected in LibreHardwareMonitor

Fraser-Watson, 2023-02-24:
I recently got a replacement Kraken x63, direct from NZXT as a RMA replacement, however this new one does not show in LibreHardwareMonitor. My old x63 is detected. I did a drop-in replacement and ensured everything was connected but it still isn't showing. Both AIOs are on the same firmware version (v2.1.0). 

I initially noticed this as FanControl was not detecting the unit, and knowing it uses LibreHardwareMonitorLib, I checked to see if LibreHardwareMonitor detects it, and it does not, at least not on the current release.

If anyone has another solution I would really appreciate it, as I need to send my old one back soon but I don't want to be stuck with an AIO I cannot fully control (I'm not installing CAM).

Thanks

brokenmass, 2023-05-18:
The issue is that new Kraken X3 have a differnt PID (0x2014) than the older (and supported) ones (0x2007)

Fraser-Watson, 2023-05-21:
> The issue is that new Kraken X3 have a differnt PID (0x2014) than the older (and supported) ones (0x2007)

Does the problem lie with LibreHardwareMonitor or FanControl?

brokenmass, 2023-05-21:
Libre hardware. I’ve raised a PR https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1078

min the meantime fancontrol users can use this plugin I wrote: 
https://github.com/brokenmass/Fancontrol.NzxtKraken

Fraser-Watson, 2023-05-21:
> Libre hardware. I’ve raised a PR #1078
> 
> min the meantime fancontrol users can use this plugin I wrote: https://github.com/brokenmass/Fancontrol.NzxtKraken

Nice. I no longer have the 0x2014 version of the x63, I just kept my current version... but this is great for anyone else who has this problem. Thank you

brokenmass, 2023-05-30:
Closed by #1078