LibreHardwareMonitor/LibreHardwareMonitor #997: Asus EC Chip is not recognized

FAaBbiii, 2023-03-06:
Hi,
I have the ROG STRIX Z790-F GAMING WIFI motherboard and unfortunately LHM does not recognize the built-in Asus EC chip and therefore also the sensors of this chip.
The AsusHW WMI interface probably no longer exists here. So I can't query the values there either.
With the current HWInfo Version 7.40 I can query the values.
[image]
[image]
I am currently using the NuGet package for a C# dashboard. Is there an update for this or an alternative way to get the sensor data?
Thx!

Rem0o, 2023-03-10:
Feel free to contribute and test:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/EC/EmbeddedController.cs

StainlessSteele, 2024-08-29:
HWInfo can see this sensor, how can we import/see what they are using and leverage it for this so we can use it for Fan Control as well?