LibreHardwareMonitor/LibreHardwareMonitor #1006: Gigabyte 4090 only showing 2 fans and 203390 RPM

Landmine-1252, 2023-03-09:
Anything I can do to resolve, what I assume are false readings?

Version: 0.9.1

[image]

PhyxionNL, 2023-03-15:
That is very fast 🚀Can you attach report? File > Save Report.

Landmine-1252, 2023-03-15:
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10983011/LibreHardwareMonitor.Report.txt)
Please see attached.

andikay, 2023-03-21:
You could also just check one of the several other programs that reads HW data, like HWMonitor, HWiNFO, GPU-Z, etc., to see if it's a software issue...? Also, you can just look at your card and listen to the fans.

Landmine-1252, 2023-03-21:
Interesting... These are showing different results at the spike. I can hear the fans ramp during the spike. I've recently attempted to solve this using FanControl and a custom fan curve.

CPUID HWMonitor
[image]

[image]

[image]

Landmine-1252, 2023-03-21:
LHM showed 30,000 RPM and HWMonitor give me 1846 RPM this go around. It spikes for less than a second so it seems like sometimes it is missed.

GPU-Z showed a max of 26,560 RPM at 77% fan control on this last test. 

[image]