LibreHardwareMonitor/LibreHardwareMonitor #1009: Gigabyte X570 Aorus Pro and other Aorus Boards no Control for Fans on IT8792E

Sn00kiT, 2023-03-11:
i know it´s a old problem but there is still no fix for it. 
The problem is that the mainboard has 2 ITE Chips (IT8688E&ITE879E) and the second one does not work. Fans are recognized but the Control does not Work and the sensor for it neither.
HWInfo shows it correctly and the fans are adjustable by Gigabyte SIV.

If have seen that there is a fork. Can you maybe merge this into LHM?
https://github.com/frankcrawford/it87/tree/master/Sensors%20configs

[image]
[image]

bearlygeek, 2023-03-11:
+1 on this request - I have the X570S Pro AX 1.1

PhyxionNL, 2023-03-15:
See #731.

closed 2023-03-15 by PhyxionNL