LibreHardwareMonitor/LibreHardwareMonitor #1019: "Max CPU/Thread Usage" equivalent

mr-swaz, 2023-03-18:
Hi, would it be possible to add an equivalent of HWiNFO's "Max CPU/Thread Usage"?
It's described as "maximum value of all current core/thread utilizations".
It's pretty useful to detect CPU bottlenecks, without having to compare all the maximum values of each thread.

[image]

MattLichter, 2023-04-10:
I would really like this too.  So much so that I already modified the source in my own fork.  Seeing as there is some interest, I'll prepare a PR and see what the devs have to say.

mr-swaz, 2023-08-11:
> I would really like this too. So much so that I already modified the source in my own fork. Seeing as there is some interest, I'll prepare a PR and see what the devs have to say.

Thanks @MattLichter, but I see that you added the new virtual sensor to GenericCpu.cs only. In fact it doesn't show up using an Intel CPU. Would it be possible to add it to IntelCpu.cs and Amd*.cs as well?

MattLichter, 2023-08-16:
@mr-swaz GenericCpu is the base class of the IntelCpu and Amd*Cpu classes.  By adding the feature to GenericCpu I have added it to every CPU class.

This feature is currently implemented in `master` but has not be released in a tagged version yet.  It will be available in the version after `0.9.2`.  Perhaps that is why you are not seeing the new behavior?

closed 2023-08-16 by mr-swaz