LibreHardwareMonitor/LibreHardwareMonitor #1021: i5-13500h The cpu cannot read the cpu frequency and temperature

lisonglin980, 2023-03-21:
i5-13500h The cpu cannot read the cpu frequency and temperature
[image]
[image]
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/11026166/LibreHardwareMonitor.Report.txt)

PhyxionNL, 2023-03-21:
Duplicate of #1021.
Status: Installing driver "E:\下载\LibreHardwareMonitor-net472\LibreHardwareMonitor.sys" failed and file exists

closed 2023-03-21 by PhyxionNL

MrLee980, 2023-03-21:
The LibreHardwareMonitor.sys file does not exist in the E:\下载\librehardwareMonitor. net472 2\ directory