LibreHardwareMonitor/LibreHardwareMonitor #1030: Sensor ITE IT8686E on a Gigabyte X370 Gaming K7 are not read correctly and no Fancontrol possible

KillerGER87, 2023-03-25:
Sensors on a Gigabyte X370 Gaming K7 are shown but i can't control any of my Fanspeeds 
Controler 1: ITE IT8686E
Controler 2 ITE IT8791E/IT8792E
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/11068646/LibreHardwareMonitor.Report.txt)

If anyone can help would be much appreciate

BenNeumeister, 2023-03-26:
You need to add the 'control' functions into the configs for your board.
Adding those might fix the secondaries too.

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/SuperIOHardware.cs#L1207

Example;
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/SuperIOHardware.cs#L1252

KillerGER87, 2023-03-26:
Thanks for your Input.
The only Problem is that i have no idea how to do that. I have no knowlege in programming.
 So can you are anyone else do that for me?

BenNeumeister, 2023-03-26:
> Thanks for your Input. The only Problem is that i have no idea how to do that. I have no knowlege in programming. So can you are anyone else do that for me?

The problem is that you will need to compile it locally anyway in order to test update, so why not learn how to do it?

Get VS2022 from here (its free);
https://visualstudio.microsoft.com/downloads/

Install the .NET framework.
You will need the net4.x, 6 and 7 runtimes.

Download GitDesktop, do a clone of this repo and give it ago.
Once it working, do a PR.

KillerGER87, 2023-03-26:
Thanks a lot! It does Controle know the first 5 Fans on the first Controler but the second one does still nothing. My question is now how can i submit my changes to here?

BenNeumeister, 2023-03-28:
Great it worked, the 2nd gigabyte controller was just added so might be some bugs in some instances.
Interesting a board so old has 8 fan controllers.

From GitDesktop you should be able to raise a PR against the main repo.

PR #1145 (2023-08-31): Add IT8686E Controls
Adds fan speed controls for AX370 Gaming K7, AX370 Gaming 5, and AB350 Gaming 3
Resolves #1030
closed 2023-08-31 by PhyxionNL