LibreHardwareMonitor/LibreHardwareMonitor #1042: NullReferenceException

stefxx, 2023-04-07:
My little VB.net program runs fine on multiple systems except one. It crashes with this error:

System.NullReferenceException: Object reference not set to an instance of an object.
at LibreHardwareMonitor.Hardware.Motherboard.Lpc.Ipmi.Update(StringBuilder stringBuilder)
at LibreHardwareMonitor.Hardware.Motherboard.Lpc.Ipmi..ctor(Manufacturer manufacturer)
at LibreHardwareMonitor.Hardware.Motherboard.Lpc.LpcIO..ctor(Motherboard motherboard)
at LibreHardwareMonitor.Hardware.Motherboard.Motherboard..ctor(SMBios smBios, ISettings settings)
at LibreHardwareMonitor.Hardware.Motherboard.MotherboardGroup..ctor(SMBios smbios, ISettings settings)
at LibreHardwareMonitor.Hardware.Computer.AddGroups()
at LibreHardwareMonitor.Hardware.Computer.Open()

Where should I start investigating? Any suggestions?

stefxx, 2023-04-07:
The impacted system is a "Supermicro X11SCL-IF" with an "Intel Xeon E-2176G" processor.

stefxx, 2023-04-08:
I've tried running the LibreHardwareMonitor program 0.9.2 and it doesn't even start on this system. It seems to die silently.

PhyxionNL, 2023-04-11:
Please run LHM under a debugger and see what's null.

doolbneerg, 2023-04-27:
I think this is the same as #1054.

closed 2023-04-28 by PhyxionNL

stefxx, 2023-05-01:
I can confirm this issue has been resolved.