LibreHardwareMonitor/LibreHardwareMonitor #1052: Close the window to tray?

Trider12, 2023-04-12:
I'd like to be able to close the window to the system tray. I don't like the current implementation of the "Minimize To Tray" feature, because clicking on the taskbar icon of a window with focus closes it to tray. This is, imho, unintuitive and very frustrating on a multi monitor setup - say, you want to bring the window into focus, but you didn't notice that it's actually on another screen and already has focus, so clicking the taskbar icon closes the window.

Trider12, 2023-08-07:
@tomoayan your suggestion is orthogonal to my issue.

Rofii, 2024-04-29:
+1

Please add a close to tray option. I want it to minimize to taskbar but close to tray, and it's not possible.