LibreHardwareMonitor/LibreHardwareMonitor #1054: lpmi NullReferenceException

Rem0o, 2023-04-18:
4/17/2023 7:10:05 PM: System.NullReferenceException: Object reference not set to an instance of an object.
at LibreHardwareMonitor.Hardware.Motherboard.Lpc.Ipmi.Update(StringBuilder stringBuilder) in C:\Users\Remi\source\repos\FanControl\LibreHardwareMonitor\LibreHardwareMonitorLib\Hardware\Motherboard\Lpc\Ipmi.cs:line 190

See https://github.com/Rem0o/FanControl.Releases/issues/1723

PhyxionNL, 2023-04-18:
This is indeed a very strange line for this error. Perhaps @tscislowski can give the build from https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1055 a try.

PhyxionNL, 2023-04-18:
Perhaps @tscislowski can give the build from https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1055 a try.

Rem0o, 2023-04-19:
@tscislowski 
Take LibreHardwareMonitor-net472 from
 https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/4736688928
 
Launch LibreHardwareMonitor itself, or take LibreHardwareMonitorLib.dll, and replace the one in FanControl with it.

doolbneerg, 2023-04-27:
I just hit this myself.  Looking at the code, it looks like there are some order of operation issues.  The library expects to be able to update the sensors before allocating the memory for the sensors. However, it also needs some other pieces to be initialized for it to pull the SDR. I am working through some changes now.

closed 2023-05-01 by PhyxionNL

Abiodun070, 2024-07-31:
_voltages.Add(RawToFloat(reading[1], sdr));

// _voltages is a read-only list
// reading is an array of byte, which would explode earlier if it was null
// sdr is a struct, which can't be null.

commit da03943f (2023-04-28): Fixes #1054