LibreHardwareMonitor/LibreHardwareMonitor #1062: Ryzen CPU Core speeds report very different clocks compared to Ryzen Master software

ccoenen, 2023-04-27:
I was wondering if my reported CPU clocks were correct, so I downloaded the AMD software "[Ryzen Master](https://www.amd.com/de/technologies/ryzen-master", and it reports very different speeds:

[image]

While LibreHardwareMonitor will always report _at least_ 3.7GHz for each core (and sometimes the boost clock if 4.6GHz), the Ryzen Master software shows _much lower_ clock speeds when my CPU is mostly idle.

ccoenen, 2023-04-27:
this is possibly the same problem as with the Ryzen 3000 series, reported in #983

ccoenen, 2023-04-27:
interestingly, windows task manager _also_ reports ~4GHz as speed

[image]

ntoff, 2023-04-28:
Ryzen Master is probably reporting the "effective" clock speeds, something hwinfo also does.

https://www.hwinfo.com/forum/threads/effective-clock-vs-instant-discrete-clock.5958/

ccoenen, 2023-04-29:
Interesting! I didn't know about that!