LibreHardwareMonitor/LibreHardwareMonitor #1072: incorrect temperatures on nct6799d (asus x670e-f)

vonvenera, 2023-05-07:
Hi, 
Libre is reporting incorrect temps for my motherboard, #3-#6 don't change at all.
I have tested with latest release and with latest nightly build, not difference.

I am prepared to do testing or provide additional info.

[image]

Crosenhain, 2023-06-12:
I have the same problem, but on a ASUS ROG CROSSHAIR X670E GENE - it uses the same NCT6799D.

Crosenhain, 2023-06-12:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1092

I imagine it can be adapted to work with other boards fairly easily.

tekert, 2023-06-25:
ITE IT8665E Sensor same thing
using http://rh-software.com/ i can see the sensor names and temps, maybe we can get the data from there,

tekert, 2023-06-26:
In the end it was a simple configuration error, made a pull request to fix it.
Temperatures from index 4 to 6 are still not readable on Ryzen mothers, happens with hwinfo and siv too.

PR #1224 (2023-12-31): Add support for ASUS ROG STRIX X670E-F GAMING WIFI
Added ASUS ROG STRIX X670E-F GAMING WIFI support to have access to T_Sensor.
Should fix #1072