LibreHardwareMonitor/LibreHardwareMonitor #1075: Invisible driver file .sys

ios1ph, 2023-05-11:
When computer.Open() there is driver file created appname.sys in exe directory.
There was a method in old HardwareMonitor to extract driver in temp folder which was better.
It is not cool when driver file extracted in exe path when you have portable app with single exe.
Could you add possibility to hide driver or choose directory to extract driver?
I mean I have already changed and created my own dll. But would be cool if you make changes here too, cuz I want to be updated using nuget as sson as new version comes out, to not every time update code by myself.

PhyxionNL, 2023-05-18:
No, it works fine in portable scenarios as the sys file should be cleaned after use.

closed 2023-05-18 by PhyxionNL

ios1ph, 2023-05-19:
It visible. When you run exe app from your desktop and sys file creates on you desktop lol.