LibreHardwareMonitor/LibreHardwareMonitor #1084: CPU package temperature Intel i12700T not showing

Coarist, 2023-05-28:
```
Hardware: 12th Gen Intel Core i7-12700T
        Sensor: CPU Core #1 Thread #1, value: 89.790085
        Sensor: CPU Core #1 Thread #2, value: 0.030398369
        Sensor: CPU Core #2 Thread #1, value: 33.4176
        Sensor: CPU Core #2 Thread #2, value: 0.030398369
        Sensor: CPU Core #3 Thread #1, value: 33.41888
        Sensor: CPU Core #3 Thread #2, value: 0.030082464
        Sensor: CPU Core #4 Thread #1, value: 33.42464
        Sensor: CPU Core #4 Thread #2, value: 0
        Sensor: CPU Core #5 Thread #1, value: 3.919357
        Sensor: CPU Core #5 Thread #2, value: 0.030720234
        Sensor: CPU Core #6 Thread #1, value: 10.0896
        Sensor: CPU Core #6 Thread #2, value: 0.030082464
        Sensor: CPU Core #7 Thread #1, value: 35.28384
        Sensor: CPU Core #7 Thread #2, value: 0.030082464
        Sensor: CPU Core #8 Thread #1, value: 1.5084803
        Sensor: CPU Core #8 Thread #2, value: 0.030082464
        Sensor: CPU Core #9, value: 34.648315
        Sensor: CPU Core #10, value: 34.012585
        Sensor: CPU Core #11, value: 33.45067
        Sensor: CPU Core #12, value: 35.44725
        Sensor: CPU Total, value: 13.934219
        Sensor: CPU Core #1, value:
        Sensor: CPU Core #2, value:
        Sensor: CPU Core #3, value:
        Sensor: CPU Core #4, value:
        Sensor: CPU Core #5, value:
        Sensor: CPU Core #6, value:
        Sensor: CPU Core #7, value:
        Sensor: CPU Core #8, value:
        Sensor: CPU Core #9, value:
        Sensor: CPU Core #10, value:
        Sensor: CPU Core #11, value:
        Sensor: CPU Core #12, value:
        Sensor: CPU Package, value:
        Sensor: CPU Core #1 Distance to TjMax, value:
        Sensor: CPU Core #2 Distance to TjMax, value:
        Sensor: CPU Core #3 Distance to TjMax, value:
        Sensor: CPU Core #4 Distance to TjMax, value:
        Sensor: CPU Core #5 Distance to TjMax, value:
        Sensor: CPU Core #6 Distance to TjMax, value:
        Sensor: CPU Core #7 Distance to TjMax, value:
        Sensor: CPU Core #8 Distance to TjMax, value:
        Sensor: CPU Core #9 Distance to TjMax, value:
        Sensor: CPU Core #10 Distance to TjMax, value:
        Sensor: CPU Core #11 Distance to TjMax, value:
        Sensor: CPU Core #12 Distance to TjMax, value:
        Sensor: Core Max, value:
        Sensor: Core Average, value:
        Sensor: CPU Core #1, value:
        Sensor: CPU Core #2, value:
        Sensor: CPU Core #3, value:
        Sensor: CPU Core #4, value:
        Sensor: CPU Core #5, value:
        Sensor: CPU Core #6, value:
        Sensor: CPU Core #7, value:
        Sensor: CPU Core #8, value:
        Sensor: CPU Core #9, value:
        Sensor: CPU Core #10, value:
        Sensor: CPU Core #11, value:
        Sensor: CPU Core #12, value:
```

Coarist, 2023-05-28:
This is because on Windows computer, elevated permission is required to access the information. In the same directory of the project executable, `app.manifest` file 

`<requestedExecutionLevel level="requireAdministrator" uiAccess="false"/>`

will prompt the user for permissions when invoking your program. Here is the results:

`
Hardware: 12th Gen Intel Core i7-12700T
        Sensor: CPU Core #1 Thread #1, value: 10.410559
        Sensor: CPU Core #1 Thread #2, value: 0
        Sensor: CPU Core #2 Thread #1, value: 12.314635
        Sensor: CPU Core #2 Thread #2, value: 0
        Sensor: CPU Core #3 Thread #1, value: 14.300358
        Sensor: CPU Core #3 Thread #2, value: 0
        Sensor: CPU Core #4 Thread #1, value: 7.386094
        Sensor: CPU Core #4 Thread #2, value: 0
        Sensor: CPU Core #5 Thread #1, value: 33.067734
        Sensor: CPU Core #5 Thread #2, value: 0
        Sensor: CPU Core #6 Thread #1, value: 14.575624
        Sensor: CPU Core #6 Thread #2, value: 0
        Sensor: CPU Core #7 Thread #1, value: 7.055706
        Sensor: CPU Core #7 Thread #2, value: 0
        Sensor: CPU Core #8 Thread #1, value: 5.764824
        Sensor: CPU Core #8 Thread #2, value: 0
        Sensor: CPU Core #9, value: 7.462448
        Sensor: CPU Core #10, value: 7.1724296
        Sensor: CPU Core #11, value: 6.740135
        Sensor: CPU Core #12, value: 6.9384394
        Sensor: CPU Total, value: 3.2954514
        Sensor: CPU Core #1, value: 51
        Sensor: CPU Core #2, value: 50
        Sensor: CPU Core #3, value: 52
        Sensor: CPU Core #4, value: 50
        Sensor: CPU Core #5, value: 53
        Sensor: CPU Core #6, value: 55
        Sensor: CPU Core #7, value: 53
        Sensor: CPU Core #8, value: 49
        Sensor: CPU Core #9, value: 52
        Sensor: CPU Core #10, value: 52
        Sensor: CPU Core #11, value: 52
        Sensor: CPU Core #12, value: 52
        Sensor: CPU Package, value: 54
        Sensor: CPU Core #1 Distance to TjMax, value: 49
        Sensor: CPU Core #2 Distance to TjMax, value: 50
        Sensor: CPU Core #3 Distance to TjMax, value: 48
        Sensor: CPU Core #4 Distance to TjMax, value: 50
        Sensor: CPU Core #5 Distance to TjMax, value: 47
        Sensor: CPU Core #6 Distance to TjMax, value: 45
        Sensor: CPU Core #7 Distance to TjMax, value: 47
        Sensor: CPU Core #8 Distance to TjMax, value: 51
        Sensor: CPU Core #9 Distance to TjMax, value: 48
        Sensor: CPU Core #10 Distance to TjMax, value: 48
        Sensor: CPU Core #11 Distance to TjMax, value: 48
        Sensor: CPU Core #12 Distance to TjMax, value: 48
        Sensor: Core Max, value: 55
        Sensor: Core Average, value: 51.75
        Sensor: CPU Core #1, value: 3949.8384
        Sensor: CPU Core #2, value: 1579.9353
        Sensor: CPU Core #3, value: 1086.2056
        Sensor: CPU Core #4, value: 987.4596
        Sensor: CPU Core #5, value: 987.4596
        Sensor: CPU Core #6, value: 987.4596
        Sensor: CPU Core #7, value: 987.4596
        Sensor: CPU Core #8, value: 987.4596
        Sensor: CPU Core #9, value: 987.4596
        Sensor: CPU Core #10, value: 987.4596
        Sensor: CPU Core #11, value: 987.4596
        Sensor: CPU Core #12, value: 987.4596
        Sensor: CPU Package, value: 10.038846
        Sensor: CPU Cores, value: 2.5956295
        Sensor: CPU Memory, value: 0
        Sensor: CPU Core, value: 0.68395996
        Sensor: CPU Core #1, value: 0.6789551
        Sensor: CPU Core #2, value: 0.6789551
        Sensor: CPU Core #3, value: 0.67529297
        Sensor: CPU Core #4, value: 0.67529297
        Sensor: CPU Core #5, value: 0.6689453
        Sensor: CPU Core #6, value: 0.6689453
        Sensor: CPU Core #7, value: 0.67626953
        Sensor: CPU Core #8, value: 0.67626953
        Sensor: CPU Core #9, value: 0.67626953
        Sensor: CPU Core #10, value: 0.67626953
        Sensor: CPU Core #11, value: 0.67626953
        Sensor: CPU Core #12, value: 0.67626953
        Sensor: Bus Speed, value: 98.74596
`

PhyxionNL, 2023-06-02:
Yes, this makes more sense to have at requireAdministrator, changed in latest commit.

closed 2023-06-02 by PhyxionNL