LibreHardwareMonitor/LibreHardwareMonitor #1086: FanControl (an application using LHM) Causing Concerning Voltage Spikes in Ryzen 7000X3D

AndyU96, 2023-05-30:
(All voltage monitoring is done with HWiNFO64)

About a week ago, I noticed that the VSOC on HWiNFO64 showed a maximum reading of 2.480V, which is 2 times the normal operating voltage that is 1.240V. The voltage spikes occurs very briefly and is pretty much only detectable at high sensor polling rates (100ms). 

After days of testing to see whether the issue was caused by BIOS or background programs, I have come to the conclusion that one of the programs I was using at the time, namely FanControl, was the culprit behind the issue.

When FanControl (the creator of which told me that it is basically an overlay of LHM) is open, at least once per hour, certain spikes occur in VSOC and VCORE simultaneously. There are also occasional above TJMAX temperature readings on the CPU Temperature sensor, along with a mind boggling minimum reading of 0 Celsius on the Max Core Temp sensor (this may be a custom sensor that I defined myself, I cant recall). The voltage spikes are not limited to being exactly double the normal amount however, I have had multiple instances of VSOC spikes at in-between voltage values, like 1.86V and 1.653V. I have a video that shows these spikes: https://youtu.be/xEVZHmkb3RU

In my video someone (who apparently doesnt use FanControl but still gets the same kind of spikes) has posted a comment saying that they've talked to Asus and HWiNFO on the issue, and consequently came to the conclusion that external software can sometimes report double values. However, as I said, with my system I have seen multiple instances of non-2x values. 

As I've subtly implied above, I got in contact with the creator of FanControl on this issue, and his reply to me was as follows:
```
Hi,
 
First, be aware that FanControl is basically an overlay on top of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor.   So everything to do with sensor readings can be found in there, FanControl uses the exact same thing. 

AFAIK FanControl/LHM does not/can’t act on the voltage of the cpu itself, it just gets readings from the LPC/superIO chips. 

Now, I never personally seen anything like this, but with quick research, I’ve seen some various weird behaviors with theses X3D chips online, see https://youtu.be/wZ-QVOKGVyM?t=143 where Asus board would fry theses chips, so I don’t know if manufacturers are still figuring out the BIOS configurations for these chips. 

Also, I’ve seen various instance where the voltage reported by third party software wasn’t reliable for AMD chips, and AMD said that you could only rely really on AMD Ryzen master software to get accurate readings. Don’t know how valid that statement is, so take it with a grain of salt.

So in order of probability for your issue, I would guess:

1) The readings don’t reflect what’s actually going on, maybe due to…. 
2) … The BIOS configs for theses chips aren’t there just yet
3) … Different monitoring software running at the same time may conflict with each other, especially with high polling rates.
 
One thing you could try though, is to disable the InpOut driver in FanControl if you got it activated (sensor settings), and see if something changes. 
 
Rémi Mercier
Fan Control software developer
```
The reason I'm creating this 'issue' post here is that I want to know whether LHM 'could' have any impact on those voltages in any way. In order to make it easier for the developer to answer this question in as clear a way as possible, I'd like to list the possible cases that come to mind, please try to either choose one of the options below that you find to be true, or if none fits what you have in mind,  I would really appreciate it if you could explain what you think might be the case, along with who I should contact about this issue instead.

Prospect A:  LHM can neither affect voltages, nor sensors, nor can it interfere with other programs' access to those sensors.
Prospect B:  LHM cannot affect voltages OR change values of sensors, but it may interfere with other programs' access to those sensors.
Prospect C:  LHM cannot affect voltages, but it may change values of some sensors at their roots or something, making them report false values.
Prospect D:  LHM may affect voltages.

Thank you in advance for any level of diligent analysis of the issue.
```
My full system specs:

Motherboard: Gigabyte B650I Aorus Ultra (Mini ITX)
CPU: Ryzen 7950X3D
RAM: G.Skill Trident Z5 6000MHz CL30 2x32GB
PSU: Cooler Master V850 SFX (v2)
GPU: MSI RTX 3080 Gaming Z Trio 10GB LHR
SSDs: Crucial P5 2TB, Samsung 980 Pro 1TB, Samsung 970 Evo Plus 2TB
HDD: Seagate 2.5" 5TB (ST5000LM000)
CPU TIM: Thermal Grizzly Kryonaut (well spread in excess)
AIO: NZXT Kraken Z63 RGB
Monitor: LG 27GN950 (using DP1.4)
Keyboard: Razer Deathstalker Pro TKL
Mouse: Logitech G303 Shroud Edition
Case: SSUPD Meshlicious

Related BIOS settings:

Version: F6a (Released 18/05/2023) [F5 tested too, same results]
EXPO: Enabled
Curve Optimizer: Per core, stability confirmed through excessive per-core and all-core OCCT testing with various instruction sets
Integrated Graphics: Disabled

Programs:

Process Lasso
FanControl
NZXT CAM
MSI Afterburner: Used for graphing sensors. Polling rate = 100ms
HWiNFO64: Provides the sensors to MSI Afterburner. Polling rate = 100ms
Steam
Discord
OpenRGB
Wallpaper Engine
ClipDiary
Razer Synapse
Imdisk (RamDisk)
```

PhyxionNL, 2023-06-02:
No, it's just reading it out.

closed 2023-06-02 by PhyxionNL

rubeste, 2023-12-30:
As far as I am aware LHM cannot change voltages/sensors. This would suggest `Prospect A`. However, this does not explain the 'Voltage spikes'. What I think happens is that LHM sometimes doubles the value as with extremally fast polling rates it could be possible that LHM (probably due to some delay a.k.a lag) combine the values of two measurements into one measurement. 

Lastly, while I can understand the frustration with putting a lot of effort for a post and getting a simple response. It is not wrong what PhyxionNL says. It is pretty direct and if you are not familiar with the lingo used, it can cause confusion. You should also take into account that a lot of open source software is maintained on voluntary basis. This means that maintainers are doing it for fun. It is not their job to play customer service. So have a little more understanding.

I hope this answers your question.