LibreHardwareMonitor/LibreHardwareMonitor #1094: GPU VRAM auto disappears in the gadget!

Omar-Os, 2023-06-12:
When I set the VRAM of the GPU to be shown in the gadget, it resets with every restart and disappears/hide and I have to open settings again to enable it! Sometimes it does that even without restarting!

SoggedFox, 2025-02-17:
Keep "GPU Bus" turned on since it also affects the visibility of "GPU Memory".

SoggedFox, 2025-02-19:
Should be under [GPU model] > Load . It's been mentioned here #971 and  #1335 .

Omar-Os, 2025-02-19:
I know, no "GPU Bus" is there.
[image]

SoggedFox, 2025-02-19:
That looks different from mine despite having the same GPU. Bus should be the second "GPU VRAM" line, try to rename and delete its name, leaving it blank, and the original name should pop up once you press enter.