LibreHardwareMonitor/LibreHardwareMonitor #1103: Lib reports no GPU sensors

TheLaGmAn, 2023-06-24:
Im using the latest lib (0.9.2) and its not reporting any GPU sensor at all (i have an nvidia 3070 ti). This was working with 0.9.0. The issue started with 0.9.1.

This is the code im using for testing

`        Dim computer = New Computer() With {
            .IsGpuEnabled = True,
            .IsControllerEnabled = True
        }

        computer.Open()
        computer.Accept(New UpdateVisitor())

        For Each hw In computer.Hardware
            log($"Hardware {hw.Name}")

            For Each shw In hw.SubHardware
                Console.WriteLine($"Sub-hardware {shw.Name}")
                For Each sensor In hw.Sensors
                    log($"  Sensor: {sensor.Name}, Value: {sensor.Value}")
                Next
            Next

            For Each sensor In hw.Sensors
                log($"  Sensor: {sensor.Name}, Value: {sensor.Value}")
            Next

            Console.WriteLine()
        Next`

The output is completly empty.
If i add
IsCpuEnabled = True, i do get CPU sensors.
If i add
IsMemoryEnabled = True, i do get mem sensors.

But the IsGpuEnabled returns nothing at all.

Any ideas?

Serafimsd, 2023-06-25:
```
For Each shw In hw.SubHardware
                Console.WriteLine($"Sub-hardware {shw.Name}")
                For Each sensor In **hw**.Sensors
```
replace to
```
For Each shw In hw.SubHardware
                Console.WriteLine($"Sub-hardware {shw.Name}")
                For Each sensor In **shw**.Sensors
```

TheLaGmAn, 2023-06-25:
Sorry about the type, it was after different tests that it was introduced but its doesnt fix the issue.
Hardware is empty so it still never reaches the typo there.
Let me clarify, here is the output of hardware in execution time

[image]

Hardware returns empty for IsGpuEnabled

TheLaGmAn, 2023-06-25:
Okay, found the partial issue, if i set the project to x64, it returns GPU sensors, but i need my project in x86.
Any idea why the project in x86 doesnt return GPU sensors?