LibreHardwareMonitor/LibreHardwareMonitor #1106: [Docs] Please improve the README and documentation

eabase, 2023-06-27:
Hi, 
Very nice project, but it need some more love for certain part of the documentation. 

* **`README`**:  
- Need a complete and functional example (not only a code snippet.)
- What are the different HW *classes* that can be monitored and what is monitored in these classes?  
  
```powershell
	$PC.IsBatteryEnabled 		= $False	# 
	$PC.IsControllerEnabled 	= $True		# ?? Fan Speed?
	$PC.IsCpuEnabled 		= $True		# Bus Speed, Frequency, Usage, Temperature, Power Draw
	$PC.IsGpuEnabled 		= $True		# Core Frequency, Memory Frequency, GPU Usage, Temperature
	$PC.IsMemoryEnabled 		= $False	# Ram usage
	$PC.IsMotherboardEnabled 	= $True		# Vcore, DRAM voltage, System Temperature, CPU Temperature
	$PC.IsNetworkEnabled 		= $False	# 
	$PC.IsPsuEnabled 		= $False	# 
	$PC.IsStorageEnabled 		= $False	# S.M.A.R.T, SSD Wear Level, Read/Write

```
The comments after `#` are my wild guesses...

PhyxionNL, 2023-07-07:
The sample works fine, copy pasting it 1:1 in a C# project will give you all hardware and sensors. Most of these properties are quite self explaining if you have any idea about the components in your PC 😁

closed 2023-07-07 by PhyxionNL

eabase, 2023-07-09:
> The sample works fine, copy pasting it 1:1 in a C# project will give you all hardware and sensors. Most of these properties are quite self explaining if you have any idea about the components in your PC 😁

This is exactly the kind of comment and behavior that allows 99% of all github projects to have 💩  documentation. You are even too lazy to edit the copy paste comments in the code itself. I was considering to help you out with some of that, but you just killed any will to help or post the solutions to several other issues here. 👎 

Good Luck.