LibreHardwareMonitor/LibreHardwareMonitor #1117: Process terminated error Server 2022

jvedman, 2023-07-14:
Windows Server 2022 21H2
Dell Poweredge R350

On starting the app:
Event ID: 1026 .NET Runtime
Application: LibreHardwareMonitor.exe Framework Version: v4.0.30319 Description: The process was terminated due to an unhandled exception. Exception Info: System.NullReferenceException at LibreHardwareMonitor.Hardware.Motherboard.Lpc.Ipmi.Update(System.Text.StringBuilder) at LibreHardwareMonitor.Hardware.Motherboard.Lpc.Ipmi..ctor(LibreHardwareMonitor.Hardware.Motherboard.Manufacturer) at LibreHardwareMonitor.Hardware.Motherboard.Lpc.LpcIO..ctor(LibreHardwareMonitor.Hardware.Motherboard.Motherboard) at LibreHardwareMonitor.Hardware.Motherboard.Motherboard..ctor(LibreHardwareMonitor.Hardware.SMBios, LibreHardwareMonitor.Hardware.ISettings) at LibreHardwareMonitor.Hardware.Motherboard.MotherboardGroup..ctor(LibreHardwareMonitor.Hardware.SMBios, LibreHardwareMonitor.Hardware.ISettings) at LibreHardwareMonitor.Hardware.Computer.set_IsMotherboardEnabled(Boolean) at LibreHardwareMonitor.UI.MainForm..ctor() at LibreHardwareMonitor.Program.Main() 

Followed by:
Event ID: 1000 Application Error
LibreHardwareMonitor.exe
0.9.2.0
a4f707d7
unknown
0.0.0.0
00000000
c0000005
00007ffaba7f5bee
5198
01d9b66de6fc402c
C:\LibreHardwareMonitor\LibreHardwareMonitor.exe
unknown
fc8f6db3-48e2-47cd-85fb-cda0490e00bf

PhyxionNL, 2023-07-15:
You need the latest build, it's already fixed there.

jvedman, 2023-07-17:
Using the latest night build link from the readme.md I am still getting the same error.