LibreHardwareMonitor/LibreHardwareMonitor #1150: System.ArgumentOutOfRangeException with BIOS Date Parsing

BoiHanny, 2023-08-29:
# Description
While using LibreHardwareMonitor, someone encountered a System.ArgumentOutOfRangeException mentions explicitly that the "Year, Month, and Day parameters describe an un-representable DateTime." 

This issue seems similar to a closed issue ([#297](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/297)), which was apparently fixed in a later update (#432). However, the problem still persists in our case.

# Details
Here is the exact stack trace of the error:
`System.ArgumentOutOfRangeException: Year, Month, and Day parameters describe an un-representable DateTime.
   at System.DateTime..ctor(Int32 year, Int32 month, Int32 day)
   at LibreHardwareMonitor.Hardware.BiosInformation.GetDate(String date)`

MySelections, 2024-01-23:
Waiting for the author's reply (#1248)

MySelections, 2024-01-23:
You could give this a shot to see if it’s sorted out.

closed 2024-07-31 by PhyxionNL