LibreHardwareMonitor/LibreHardwareMonitor #1151: No Temperature reading for Samsung 970 EVO+ NVMe SSDs with LHM 0.9.1+

andikay, 2023-08-29:
I have been using LHM 0.9.0 for a while now and I love it (except for the lack of a built-in dark mode, but you can't have everything I suppose :D). However, whenever I try to update it (i.e. extracting the zip into a new folder, not using the config from the old version, etc.) I see that every version after 0.9.0 is missing the Temperature reading for my Samsung NVMe SSDs. Something must have gone missing from 0.9.0 to 0.9.1.

This is how it looks in 0.9.0, no sensors are hidden:
[image]

And this is the current 0.9.2 (but it looks the same in 0.9.1, I tested it), no sensors are hidden:
[image]

sebastian-dev, 2023-09-05:
I also have an Evo970 Plus with 2TB installed and all temperatures are read out. (Windows 10 and 11)

andikay, 2023-09-05:
Interesting. The system I made the screenshot on is Win7 and uses the Samsung NVMe driver. I also use the program on Win10 but I haven't tested any other version than 0.9.0 for it so far. Nonetheless, it's working on 0.9.0 and not working on follow-up versions on Win7, so something must have been changed or removed in that regard, which I find unfortunate.

sebastian-dev, 2023-09-06:
Windows 7 is obsolete and not supportet anymore by Micosoft.
I and many others no longer test anything for Windows 7 compatibility.
Windows 7 has no support since 2020.