LibreHardwareMonitor/LibreHardwareMonitor #1155: No access to temperatures of Core i7-12700H neither using the stable build or nightly build

PatZer0, 2023-08-31:
CPU clocks and temperatures are all blank, aleady running in Administrator mode.
I have tried both the 0.9.2 stable release or the latest nightly build, none of them worked.
Laptop model: Lenovo Legion Y9000P 2022 (12700H+3060)
Report.txt and CPU-Z screenshot uploaded.

Could this problem caused by software conflict?

[Uploading LibreHardwareMonitor.Report.txt…]()
[image]
[image]

PatZer0, 2023-08-31:
it seems like the report.txt was not uploaded correctly
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/12481817/LibreHardwareMonitor.Report.txt)

MaxVonEvil, 2024-01-25:
Hi, seeing the same on v0.93 running on a Lenovo T560 (Intel Core i5-6200U)
Anything we can do to help troubleshoot this?

[image]

[image]

closed 2025-02-17 by PatZer0