LibreHardwareMonitor/LibreHardwareMonitor #1157: No fan speeds in a ThinkPad P1 G5

ygoe, 2023-08-31:
On my ThinkPad P1 Gen 5 no fans were detected. I'm using the FanControl app to manage this. It works fine on my desktop computer but not the laptop. I wanted to try it out there because the embedded fan control is very annoying, turning up to full speed every few minutes, for just 2 seconds, then going down again, all with no CPU activity. I believe that's not necessary.

moaltmann, 2024-02-18:
Same issue on my ThinkPad X1 Carbon Gen 8 (20U90006GE) and also on Lenovo Thinkpad X1 Yoga 2 Gen:

FanControl via https://getfancontrol.com
[image]

Fan Control v1.6.8 via https://github.com/lich426/FanCtrl
[image]

MarkLily, 2025-04-14:
My Lenovo Thinkpad X1 also cannot detect motherboard temperature