LibreHardwareMonitor/LibreHardwareMonitor #1162: Support for ARM64

almodad, 2023-09-07:
Are there plans to support ARM? The following exception is thrown on Surface Pro X:
Unhandled exception. System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at LibreHardwareMonitor.Hardware.Cpu.CpuId..ctor(Int32 group, Int32 thread, GroupAffinity affinity)
   at LibreHardwareMonitor.Hardware.Cpu.CpuId.Get(Int32 group, Int32 thread)
   at LibreHardwareMonitor.Hardware.Cpu.CpuGroup.GetProcessorThreads()
   at LibreHardwareMonitor.Hardware.Cpu.CpuGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.AddGroups()
   at LibreHardwareMonitor.Hardware.Computer.Open()

md2k, 2025-09-19:
@PhyxionNL  any chances it will resolved since #984 merged ?

Taiizor, 2025-10-03:
> [@PhyxionNL](https://github.com/PhyxionNL) any chances it will resolved since [#984](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984) merged ?

Honestly, I’m curious about this too, but I don’t have an ARM64 device to test it on.