LibreHardwareMonitor/LibreHardwareMonitor #1166: .NET exception on OS startup

shalak, 2023-09-09:
Recently I'm getting the .NET exception after windows boot. I have LHW added to startup. The detail log mentions:

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
   at System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   at System.Collections.Generic.List`1.get_Item(Int32 index)
   at Aga.Controls.Tree.TreeNodeAdv.get_NextNode() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:line 252
   at Aga.Controls.Tree.TreeNodeAdv.get_NextVisibleNode() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:line 281
   at Aga.Controls.Tree.TreeViewAdv.<get_VisibleNodes>d__256.MoveNext() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.Properties.cs:line 74
   at Aga.Controls.Tree.TreeViewAdv.CreateRowMap() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 890
   at Aga.Controls.Tree.TreeViewAdv.UnsafeFullUpdate() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 684
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Assembly Version: 0.9.2.0
    Win32 Version: 0.9.2.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9172.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Assembly Version: 1.7.0.0
    Win32 Version: 1.7.0.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Assembly Version: 0.9.2.0
    Win32 Version: 0.9.2.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/LibreHardwareMonitorLib.DLL
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly Version: 2.10.1.0
    Win32 Version: 2.10.1.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
OxyPlot
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/OxyPlot.DLL
----------------------------------------
OxyPlot.WindowsForms
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/OxyPlot.WindowsForms.DLL
----------------------------------------
System.Configuration.Install
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll
----------------------------------------
1ercfohj
    Assembly Version: 0.0.0.0
    Win32 Version: 4.8.9172.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
HidSharp
    Assembly Version: 2.1.0.0
    Win32 Version: 2.1.0.0
    CodeBase: file:///C:/Program%20Files/LibreHardwareMonitor-net472/HidSharp.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

Note that I do not have a `D:\` drive

seagull, 2023-09-20:
0.9.2 seems to have lots of .NET issues, but from what I've seen they are fixed in the nightlies.
We just need to wait for 0.9.3 to come out with these fixes included.

DHCanB, 2023-09-29:
Could this be a race condition? 
The validity check of the index is reasonable, but if _parent.Nodes (or _parent) could be changed from a different thread, the check would not be enough.

```
public TreeNodeAdv NextNode
{
	get
	{
		if (_parent != null)
		{
			int index = Index;
			if (index < _parent.Nodes.Count - 1)
				return _parent.Nodes[index + 1];
		}
		return null;
	}
}
```

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/a8975f5d5446b88c7c7ad3c2a2929c2c3c6c1c70/Aga.Controls/Tree/TreeNodeAdv.cs#L242C10-L242C22

Sab44, 2024-04-16:
Having the same issue, after cold boot seeing this error pop-up during start-up.
It's only happening occasionally though, which might support the race condition argument.

EyalRotem, 2024-09-07:
using version 0.9.3, still have this issue.

my versions are a bit different, but the exception is basically the same
https://pastebin.com/Q4DNWXqc

PhyxionNL, 2024-09-20:
Dupe #688

closed 2024-09-20 by PhyxionNL