LibreHardwareMonitor/LibreHardwareMonitor #1177: AMD GPU Sensors not detected after Adrenaline 23.10.1 update

Alcolawl, 2023-10-11:
After updating to to AMD Software: Adrenalin Edition 23.10.1 and restarting, FanControl can no longer detect any sensors on my AMD GPU (RX 7900XTX). I have tried re-detecting sensors, re-installing FanControl, re-installing the graphics driver, all to no avail. Other programs like HWinfo and GPU-Z detect the sensors just fine.

closed 2023-10-11 by Alcolawl

cromefire, 2023-10-20:
Actually having the same problem, did you manage to fix it or why did you close it?

Alcolawl, 2023-10-20:
> Actually having the same problem, how did you manage to fix it or why did you close it?

My reasons for closing it were two-fold:

1. Embarrassingly, I was having trouble with the sensors not showing up in another program, Fan Control, which relies on LibreHardwareMonitor. So, I guess opening an issue here wasn't completely off-base but I meant to open it on the Fan Control repo.
2. I rectified the issue by using DDU and re-installing the drivers a second time. Makes no sense, really, but it worked. First time I've ever seen this problem occur.