LibreHardwareMonitor/LibreHardwareMonitor #1184: Multiple entries on system tray in Windows

h2oboi89, 2023-10-22:
[image]

Over time the temperature icon I set in the system tray gets hidden and it seems a new one is created somehow. After a while most of the entries in "Settings -> Personalization -> Taskbar -> Other system tray icons" is filled with now dead instances of LibreHardwareMonitor.

SnDream, 2024-11-06:
[image]
Same issue. Does anyone have a solution or a way to work around it?

thx538, 2025-01-23:
Same issue here after a fresh Windows 11 installation.

[image]

mdcollins05, 2025-03-04:
I, too, have this issue. It doesn't only coincide with a reboot or restart of LibreHardwareMonitor. It seems to happen even after starting up from hibernating.