LibreHardwareMonitor/LibreHardwareMonitor #1185: LHM frequently gets flagged as a cheat program by EAC (Easy Anti-Cheat)

SoraKagami, 2023-10-24:
### LibreHardwareMonitor frequently, but not consistently gets flagged as a cheat program/exploit by Epic Game's Anti-Cheat software "Easy Anti-Cheat" (EAC).

https://forums.bhvr.com/dead-by-daylight/discussion/348223/pc-libre-hardware-monitor-occasionally-gets-flagged-by-anti-cheat-software-crash

This also happens on 7DaysToDie as of Alpha 21.1 (or was it A21.2?), also running EAC.

I have tried reporting this issue to Epic's EAC support directly twice and reported through Dead-By-Daylight's forums (ignored/rejected by staff that did not understand the issue).

**The only other way to get this resolved would be for someone in the LibreHardwareMonitor dev team to contact Epic's EAC division directly, or for someone to figure out how to prevent LHM from being flagged as a cheat program.**

The issue is most likely related to (LHM's) WinRing0 not being whitelisted.

**Known similar issues:**
* Gigabyte's RGB-Fusion (WinRing0) [here](https://www.reddit.com/r/gigabytegaming/comments/af4d06/fusion_rgb_and_anti_cheat_software/) & [here](https://steamcommunity.com/app/301520/discussions/5/1729827777333836211/)
* ASUS's RGB lighting software (WinRing0) [here](https://rog-forum.asus.com/t5/asus-software/aura-sync-trips-easy-anti-cheat/td-p/867704), [here](https://rog-forum.asus.com/t5/asus-software/lightingservice-exe-is-activating-eac-anti-cheat-on-games/td-p/802521) & [here](https://rog-forum.asus.com/t5/asus-software/lightingservice-exe-is-activating-eac-anti-cheat-on-games/td-p/802521). Does not appear resolved as they did not appear to have gotten their WinRing0 whitelisted by EAC.

SoraKagami, 2023-10-24:
### I happened upon another software out there called "ThrottleStop" that uses WinRing0 and used to be blocked by anti-cheat programs such as EAC.

**The solution the dev implemented was to remove the memory read/write capabilities from WinRing0.**
https://www.techpowerup.com/forums/threads/throttlestop-winring0x64-sys-vulnerability.266500/#post-4255585

As far as I can tell from the source code, LibreHardwareMonitor does not need to have memory-writing capabilities.
Memory writing code appears in two places, both unused:
- `LibreHardwareMonitorLib\Hardware\InpOut.cs`
- `LibreHardwareMonitorLib\WinRing0\OpenLibSys.c`

If these sections are commented out (_but left as commented-out for anyone that wants to re-add it back for custom builds_) this should remove what is the most likely cause of Anti-cheat programs flagging LibreHardwareMonitor as a cheat/hack program. **Not just non-referenced in code, but completely commented out from release builds so the methods/functions are not accessible by other programs.**

Ideally memory address reading capabilities should be removed too if possible, however `InpOut.cs`'s ReadMemory appears to be in use.

Rem0o, 2023-10-25:
InpOut is optional in FanControl, and is disabled by default, so the driver won't ever be loaded if you want.

As for WinRing0, see https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984

The TLDR, the current Microsoft restrictions exclude the indie dev from signing a driver.

lane-ftw, 2024-09-11:
Fancontrol V201 being flagged as cheat by EAC.
https://github.com/Rem0o/FanControl.Releases/issues/2104#issuecomment-2342410000

nmsu-adavis, 2024-11-11:
LHM is still being occasionally flagged by EAC, causing applications to force-close. This is occurring using LibreHardwareMonitorLib.Hardware in a C# application. I am unsure which specific portion of the Hardware namespace is causing the issue, though this is the only namespace in LibreHardwareMonitorLib being used in the application.

Aytackydln, 2024-11-11:
Probably WinRing0. I disabled CPU & GPU sensors for AMD users by default for Aurora users

Knewest, 2025-01-11:
Just happened to me while playing Fortnite.
[image]

PhyxionNL, 2025-03-19:
Dupe of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984.

closed 2025-03-19 by PhyxionNL