LibreHardwareMonitor/LibreHardwareMonitor #1214: Does not support Intel Arc GPU

courierkiller, 2023-12-13:
It detects the inbuilt Intel UHD Graphics 770 of my core i9 13900 but it does not detect my dedicated GUNNIR Intel Arc A770 Photon 16G OC Graphics Card.

Can you make it detect Intel Arc GPUs, please?

Thanks in advance

luser, 2024-05-03:
I did a little bit of investigation and found that Intel provides their [Intel Graphics Control Library](https://github.com/intel/drivers.gpu.control-library). For example, [ctlEnumTemperatureSensors](https://intel.github.io/drivers.gpu.control-library/Control/api.html#ctlenumtemperaturesensors) to enumerate temperature sensors on a display adapter and [ctlTemperatureGetState](https://intel.github.io/drivers.gpu.control-library/Control/api.html#ctltemperaturegetstate) to read the temperature from a specific sensor.

There are headers for the C API in the GitHub repository. However, one thing I'm unsure of is license compatibility. That repository does not contain an open source license, the `LICENSE` file is merely an "Intel Software License Agreement". The headers there define the API to the non-open-source library Intel ships with their GPU drivers. IANAL, but I *think* that using those APIs in LibreHardwareMonitor should be OK, since that license is primarily concerned with redistributing the contents of that repository.

zzzoom, 2025-09-03:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1822