LibreHardwareMonitor/LibreHardwareMonitor #1236: librehardwaremonitor VS openhardwaremonitor

NolonQ, 2024-01-06:
What EXACTLY is the difference between openhardwaremonitor and librehardwaremonitor?
Judging from the GUI it looks 1:1 the same, does it not?

jeffdamp-wave, 2024-01-11:
Looks like openhardwaremonitor is both the monitor and hardware interface in one .exe. and hasn't been updated in 4 years. librehardwaremonitor splits out the library and .exe. Other programs such as FanControl use this repos library.

NolonQ, 2024-01-13:
> Looks like openhardwaremonitor is both the monitor and hardware interface in one .exe. and hasn't been updated in 4 years. librehardwaremonitor splits out the library and .exe. Other programs such as FanControl use this repos library.

Thank you, so OHM is simply outdated then i'll stick to LHW, thanks again.