LibreHardwareMonitor/LibreHardwareMonitor #1240: CPU and GPU Power usage aren't here, using CS program

2DotWasHere, 2024-01-17:
I've create an CS program to track CPU and GPU power usage, using example from readme, but for some reason it won't report those stats, even tho i have them in official `LibreHardwareMonitor.exe`.

Here's my code: [link](https://gist.github.com/2DotWasHere/367ab142cbbafd313b2b4dc9e65c7e23)
Data i've got via `computer.GetReport()`
[Uploading hardware_report.txt…]()

in app.manifest i have:
```.manifest
<requestedExecutionLevel level="requireAdministrator" uiAccess="false" />/>
```

## So do i missed anything that can block me from getting those stats?

2DotWasHere, 2024-01-17:
i found the issue, you need to build your program for x64

closed 2024-01-17 by 2DotWasHere

yarodevuci, 2024-02-08:
@DoubleIsNaN did not help for me