LibreHardwareMonitor/LibreHardwareMonitor #1255: The CPU temperature of i9-10900 cannot be recognized.

LaoHu0329, 2024-01-31:
I am using the latest i9-10900 CPU device, when I use HWiNFO64 I can detect the CPU temperature smoothly, but it is not recognized on Libre Hardware Monitor, only "-" is displayed, is there any solution, I want to use Libre Use it on Hardware Monitor. Thank you in advance.

nona00, 2024-03-15:
same issue here with i5-10500 and Windows 10.

Z3r10, 2024-06-13:
I had the same problem (Surface Pro 7), Temp / Freq stopped working (I had it before though)
It was due to Windows Defender blocking the loading of the driver (LibreHardwareMonitor.sys).

Try to add LibreHardwareMonitor folder to Exclusions:

[image]

closed 2026-01-03 by PhyxionNL