LibreHardwareMonitor/LibreHardwareMonitor #1260: Reading the library via PowerShell

Lifailon, 2024-02-04:
Good afternoon!

I'm writing a module for PowerShell (repository: https://github.com/Lifailon/PowerShellHardwareMonitor) and I have a problem when reading data directly from the library (LibreHardwareMonitorLib.dll). On two different machines, I'm not getting readings from all the sensors. If I use the CIM (Common Information Model), I get all sensors (there are many more than OpenHardwareMonitor). In the same way I can read the data of all sensors via REST API. All workarounds need a working instance of the application, but I would like to read from the library.

Can you suggest what the problem is related to? Thank you!

Examples:

```PowerShell
        $path = "$home\Documents\LibreHardwareMonitor"
        $Computer = New-Object -TypeName LibreHardwareMonitor.Hardware.Computer
        $Computer.IsCpuEnabled         = $true
        $Computer.IsGpuEnabled         = $true
        $Computer.IsMemoryEnabled      = $true
        $Computer.IsMotherboardEnabled = $true
        $Computer.IsNetworkEnabled     = $true
        $Computer.IsPsuEnabled         = $true
        $Computer.IsStorageEnabled     = $true
        $Computer.IsControllerEnabled  = $true
        $Computer.IsBatteryEnabled     = $true
        $Computer.Open()
        $Sensors  = $computer.Hardware.Sensors | Select-Object @{
            name = "HardwareName"
            expression = {
                $_.Hardware.Sensors[0].Hardware.Name
            }
        },
        @{name = "SensorName";expression = { $_.Name }},
        @{name = "SensorType";expression = { "$($_.SensorType) $($_.Index)" }},
        @{name = "Value";expression = { [int]$_.Value }},
        @{name = "Min";expression = { [int]$_.Min }},
        @{name = "Max";expression = { [int]$_.Max }}
        $Sensors | Sort-Object HardwareName,SensorType,SensorName | Format-Table
```

Output:

```
HardwareName                      SensorName                     SensorType     Value  Min  Max
------------                      ----------                     ----------     -----  ---  ---
12th Gen Intel Core i7-1260P      Bus Speed                      Clock 0          100  100  100
12th Gen Intel Core i7-1260P      CPU Core #1                    Clock 1          399  399  399
12th Gen Intel Core i7-1260P      CPU Core #10                   Clock 10        1098 1098 1098
12th Gen Intel Core i7-1260P      CPU Core #11                   Clock 11         699  699  699
12th Gen Intel Core i7-1260P      CPU Core #12                   Clock 12         599  599  599
12th Gen Intel Core i7-1260P      CPU Core #2                    Clock 2         3794 3794 3794
12th Gen Intel Core i7-1260P      CPU Core #3                    Clock 3         1697 1697 1697
12th Gen Intel Core i7-1260P      CPU Core #4                    Clock 4          599  599  599
12th Gen Intel Core i7-1260P      CPU Core #5                    Clock 5         2895 2895 2895
12th Gen Intel Core i7-1260P      CPU Core #6                    Clock 6         2795 2795 2795
12th Gen Intel Core i7-1260P      CPU Core #7                    Clock 7          899  899  899
12th Gen Intel Core i7-1260P      CPU Core #8                    Clock 8          998  998  998
12th Gen Intel Core i7-1260P      CPU Core #9                    Clock 9          899  899  899
12th Gen Intel Core i7-1260P      CPU Total                      Load 0             4    4    4
12th Gen Intel Core i7-1260P      CPU Core Max                   Load 1            99   99   99
12th Gen Intel Core i7-1260P      CPU Core #5                    Load 10           15   15   15
12th Gen Intel Core i7-1260P      CPU Core #6                    Load 11           15   15   15
12th Gen Intel Core i7-1260P      CPU Core #7                    Load 12           15   15   15
12th Gen Intel Core i7-1260P      CPU Core #8                    Load 13           21   21   21
12th Gen Intel Core i7-1260P      CPU Core #9                    Load 14           15   15   15
12th Gen Intel Core i7-1260P      CPU Core #10                   Load 15           15   15   15
12th Gen Intel Core i7-1260P      CPU Core #11                   Load 16           15   15   15
12th Gen Intel Core i7-1260P      CPU Core #12                   Load 17           94   94   94
12th Gen Intel Core i7-1260P      CPU Core #1 Thread #1          Load 2            99   99   99
12th Gen Intel Core i7-1260P      CPU Core #1 Thread #2          Load 3             0    0    0
12th Gen Intel Core i7-1260P      CPU Core #2 Thread #1          Load 4            19   19   19
12th Gen Intel Core i7-1260P      CPU Core #2 Thread #2          Load 5             0    0    0
12th Gen Intel Core i7-1260P      CPU Core #3 Thread #1          Load 6            27   27   27
12th Gen Intel Core i7-1260P      CPU Core #3 Thread #2          Load 7             0    0    0
12th Gen Intel Core i7-1260P      CPU Core #4 Thread #1          Load 8            13   13   13
[... 262 lines cut ...]
Radmin VPN                       Data Uploaded                  Data 2              0     0         0
Radmin VPN                       Data Downloaded                Data 3              0     0         0
Radmin VPN                       Network Utilization            Load 1              0     0         0
Radmin VPN                       Upload Speed                   Throughput 7        0     0      3414
Radmin VPN                       Download Speed                 Throughput 8        0     0       152
vEthernet (Default Switch)       Data Uploaded                  Data 2              0     0         0
vEthernet (Default Switch)       Data Downloaded                Data 3              0     0         0
vEthernet (Default Switch)       Network Utilization            Load 1              0     0         0
vEthernet (Default Switch)       Upload Speed                   Throughput 7        0     0    147985
vEthernet (Default Switch)       Download Speed                 Throughput 8        0     0         0
WD PC SN740 SDDPNQD-1T00-1027    Data Read                      Data 4          10469     0     10469
WD PC SN740 SDDPNQD-1T00-1027    Data Written                   Data 5          10325     0     10325
WD PC SN740 SDDPNQD-1T00-1027    Available Spare                Level 1           100     0       100
WD PC SN740 SDDPNQD-1T00-1027    Available Spare Threshold      Level 2            10     0        10
WD PC SN740 SDDPNQD-1T00-1027    Percentage Used                Level 3             0     0         0
WD PC SN740 SDDPNQD-1T00-1027    Used Space                     Load 0             52    52        53
WD PC SN740 SDDPNQD-1T00-1027    Read Activity                  Load 31             0     0       100
WD PC SN740 SDDPNQD-1T00-1027    Write Activity                 Load 32             1     0       100
WD PC SN740 SDDPNQD-1T00-1027    Total Activity                 Load 33             1     0       100
WD PC SN740 SDDPNQD-1T00-1027    Temperature                    Temperature 0      46     0        68
WD PC SN740 SDDPNQD-1T00-1027    Temperature 1                  Temperature 6      56     0        87
WD PC SN740 SDDPNQD-1T00-1027    Temperature 2                  Temperature 7      46     0        68
WD PC SN740 SDDPNQD-1T00-1027    Read Rate                      Throughput 34       0     0 407017824
WD PC SN740 SDDPNQD-1T00-1027    Write Rate                     Throughput 35   12168     0 766726400
```

IvanWongw, 2024-05-17:
Is there any way to retrieve the motherboard temperature using PowerShell, as the first script failed to do so while the program running outputs the motherboard temperature?

Lifailon, 2024-05-17:
> Is there any way to retrieve the motherboard temperature using PowerShell, as the first script failed to do so while the program running outputs the motherboard temperature?

This can be done via a running instance of the LibreHardwareMonitor application, here is an example on my system:

Get-Sensor | Where-Object HardwareName -like *nuvoton* | Where-Object SensorType -like *temp* | Format-Table

HardwareName     SensorName SensorType    Value Min Max
------------     ---------- ----------    ----- --- ---
Nuvoton NCT6687D CPU        Temperature 0    38  30  60
Nuvoton NCT6687D System     Temperature 1    44  36  48
Nuvoton NCT6687D VRM MOS    Temperature 2    47  40  56
Nuvoton NCT6687D PCH        Temperature 3    48  39  53
Nuvoton NCT6687D CPU Socket Temperature 4    40  32  48
Nuvoton NCT6687D PCIe x1    Temperature 5    29  29  29
Nuvoton NCT6687D M2_1       Temperature 6    23  23  24

Lifailon, 2024-05-17:
The problem is that this data can only be retrieved through the **WMI/CIM system or REST API**, but it is impossible to retrieve all the data through the *library*, even using latency (regardless of whether the application process is running or not).