LibreHardwareMonitor/LibreHardwareMonitor #1265: No temperature reading for Atom C3758 CPU

andonovski, 2024-02-07:
I have Chinese Qotom mini-PC with Intel Atom C3758R CPU. There are no readings for CPU temp and clocks. I don't know if it's issue with motherboard or CPU, the LibeHardwareMonitor doesn't see sensors because of specific motherboard or there is no support for Atom C3000 series CPUs.
As a side note, using CoreTemp I see CPU temps. But I would like to user LibreHardwareMonitor because there are more sensors.

Any info regarding this, maybe?

Z3r10, 2024-06-13:
I had the same problem (Surface Pro 7), Temp / Freq stopped working (I had it before though)
It was due to Windows Defender blocking the loading of the driver (LibreHardwareMonitor.sys).

Try to add LibreHardwareMonitor folder to Exclusions:

[image]

andonovski, 2024-06-13:
Yep, this solved the problem. Strangely enough, on another machine it works without this exclusion! Same Windows configuration, same OS, let's say 99% are exactly the same on both machines but here I need exclusion. Really strange, but ok...

Thanks for the help!