LibreHardwareMonitor/LibreHardwareMonitor #1285: Incorrect GPU Load and Core Frequency values (Radeon RX 7900XT)

Schuischta, 2024-03-08:
I discovered that the GPU Load and Core Frequency on the RX 7900 XT (and probably other newer GPU's too) jumps randomly up and down, so it doesn't show the correct values. I saw that while I was testing my little widget-like performance monitor. -> [https://github.com/schuischta/systemloadtracker](url)

pepitomiaumiau, 2024-03-18:
Yup, can confirm. The values for my 7900XT are absolutely all over the place, most of the time jumping from 0% to 100% out of nowhere. 

The wrong values are coming from LibreHardwareMonitor, you can use it separately to test it.

epinter, 2024-04-07:
"Powers/GPU Package" has a huge variation, while amd overlay and GPU-z stays stable at 113w-116w for example, librehardwaremonitor jumps from 60 to 150 and back to lower than 100w in a second. Same happens with "Load/GPU Core" and "Load/GPU Memory".  "Clocks/GPU Core" also has a variation, but looks different.

closed 2024-04-09 by PhyxionNL