LibreHardwareMonitor/LibreHardwareMonitor #1310: Laptop CPU (AMD Ryzen * 7***U) weird values for clock speed (and Voltages)

Eno77, 2024-04-10:
Since the similar 5**** Series seem to work it may be an easy fix?

[image]

maybe due to wrong factor recognition?

[image]

And here the voltages (maybe correct, but negative values seem weird:

[image]

Eno77, 2024-04-10:
I dont know, if its related, but voltages for integrated graphics are too high too: (maybe worth another issue?)

[image]

ntoff, 2024-05-07:
Possible duplicate of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/566 Do you have core isolation turned on in windows device security settings?

Eno77, 2024-05-08:
hi @ntoff

thanks for pointing to this, you are right, unfortunately I can not deactivate the according setting since its set by the domain admin and probably should stay this way to meet security standards.

But I reproduced the issue on my private Desktop PC (5800X3D) -> turning on kernel isolation causes this, so its in fact a duplicate.

closed 2024-05-08 by Eno77