LibreHardwareMonitor/LibreHardwareMonitor #1321: .NET Exception after logon in SystemTray.Redraw()

cristianst85, 2024-04-26:
Occasionally I am seeing this exception at startup. 
Version 0.9.3 build from the latest commit 4d6a755c115b8dc497166088f2fe6db6206f6323.
Windows 10 version 22H2.

Exception details below.
```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.InvalidOperationException: Collection was modified; enumeration operation may not execute.
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   at LibreHardwareMonitor.UI.SystemTray.Redraw() in D:\Repositories\LibreHardwareMonitor\src\LibreHardwareMonitor\UI\SystemTray.cs:line 102
   at LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e) in D:\Repositories\LibreHardwareMonitor\src\LibreHardwareMonitor\UI\MainForm.cs:line 763
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9232.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Assembly Version: 0.9.3.0
    Win32 Version: 0.9.3.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9195.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9236.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Assembly Version: 1.7.0.0
    Win32 Version: 1.7.0.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Assembly Version: 0.9.3.0
    Win32 Version: 0.9.3.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/LibreHardwareMonitorLib.DLL
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly Version: 2.10.1.0
    Win32 Version: 2.10.1.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9236.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
OxyPlot
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/OxyPlot.DLL
----------------------------------------
OxyPlot.WindowsForms
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/OxyPlot.WindowsForms.DLL
----------------------------------------
System.Configuration.Install
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll
----------------------------------------
x1u3snwf
    Assembly Version: 0.0.0.0
    Win32 Version: 4.8.9236.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
HidSharp
    Assembly Version: 2.1.0.0
    Win32 Version: 2.1.0.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/HidSharp.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

EhudVardi, 2024-09-12:
I have something similar.
At windows startup it generates the exception but if i wait for a minute or so and reopen the app, then it doesn't appear.

feeding chatgpt the exception, it tells me its probably an unsafe multithread accessing the same resource without proper synch

this is my exception data 

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.InvalidOperationException: Collection was modified; enumeration operation may not execute.
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   at LibreHardwareMonitor.UI.SystemTray.Redraw() in D:\Repositories\LibreHardwareMonitor\src\LibreHardwareMonitor\UI\SystemTray.cs:line 102
   at LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e) in D:\Repositories\LibreHardwareMonitor\src\LibreHardwareMonitor\UI\MainForm.cs:line 763
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9232.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Assembly Version: 0.9.3.0
    Win32 Version: 0.9.3.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9195.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9236.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Assembly Version: 1.7.0.0
    Win32 Version: 1.7.0.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Assembly Version: 0.9.3.0
    Win32 Version: 0.9.3.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/LibreHardwareMonitorLib.DLL
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly Version: 2.10.1.0
    Win32 Version: 2.10.1.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9236.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
OxyPlot
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/OxyPlot.DLL
----------------------------------------
OxyPlot.WindowsForms
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/OxyPlot.WindowsForms.DLL
----------------------------------------
System.Configuration.Install
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll
----------------------------------------
x1u3snwf
    Assembly Version: 0.0.0.0
    Win32 Version: 4.8.9236.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
HidSharp
    Assembly Version: 2.1.0.0
    Win32 Version: 2.1.0.0
    CodeBase: file:///C:/Programs/OpenHardwareMonitor/HidSharp.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

cristianst85, 2024-09-12:
The fix should be easy to implement. The challenge is to be able to reproduce the issue consistently to check whether the fix is working properly or not.

EhudVardi, 2024-09-12:
> The fix should be easy to implement. The challenge is to be able to reproduce the issue consistently to check whether the fix is working properly or not. We don't need ChatGPT to tell us what the cause is.

you are correct, the reproduction is the key, which is a problem at the moment. I've built the solution and attempted to recreate but no luck.  chatgpt for me is just another useful tool in my toolbox. I see no reason why one should not use all tools available.

closed 2026-10-12 by Blacktempel